pauldemarco/flutter_blue #657: Empty ServiceUUID crashes discoverServices - fix included

endrelovas, 2020-08-15:
Some devices return empty strings as their ServiceUUID when I execute discoverServices. This problem is on Android only.

As a workaround replace _BluetoothCharacteristic.fromProto_ in **bluetooth_device.dart**:

```
  BluetoothCharacteristic.fromProto(protos.BluetoothCharacteristic p, {String serviceUuid = ""})
      : uuid = new Guid(p.uuid),
        deviceId = new DeviceIdentifier(p.remoteId),
        serviceUuid = new Guid(p.serviceUuid == "" ? serviceUuid : p.serviceUuid),
        secondaryServiceUuid = (p.secondaryServiceUuid.length > 0)

``` 

And inside **bluetooth_services.dart**:
```
characteristics = p.characteristics
            .map((c) => new BluetoothCharacteristic.fromProto(c, serviceUuid : p.uuid))
```

Goog luck.

BananaNeil, 2021-02-08:
I am also experiencing this on macos.

eriknorth, 2021-04-14:
Did anyone manage to solve this issue? This issue does not seem to be present on iOS.

eriknorth, 2021-04-14:
I did some more investigation and found the culprit but I do not know what is the right way to solve this and why exactly this goes wrong.

In iOS BLE services are mostly reported as Primary `isPrmary == true` (and generally it is case for the most of the BLE devices as Secondary services do not have big use), however on MacOS the very same services are reported as Secondary `isPrimary == false`. So this is the beginning of the issue.

On the flutter side the problem is that `BluetoothCharacteristic` is expected to always be part of a Primary service and that is not the case, it can be part of only Secondary service as well.

Will have a look more tomorrow, especially why MacOS would report service as Secondary if it is Primary.

Cheers.

eriknorth, 2021-04-15:
I did some more testing and the problem is what I had mentioned - MacOS for some reason does not report services as Primary but just Secondary. This seems to be a bug on Apple side. I hacked a way for services to be considered as Primary and then it started to work, however that is not even close to a clean solution. Will try to put a ticket in Apple Dev Forum about this.

Mirageofmage, 2021-05-17:
> I did some more testing and the problem is what I had mentioned - MacOS for some reason does not report services as Primary but just Secondary. This seems to be a bug on Apple side. I hacked a way [...]

Has there been any update? In the meantime can you describe the workaround you were able to use?

eriknorth, 2021-05-19:
This is a bit dirty hack and dirty diff. What I did was to remove the checks for 'isPrimary' in some places. It removes support for secondary services, however in my experience, I have not seen a device that would use a secondary service. And I use this code just because it is easier and faster to test on MacOS instead of iOS, not really planning to make a MacOS version of the app.

```
280,282c280
<     NSLog(@"locateCharacteristic primary %@", [primaryService.UUID fullUUIDString]);
<   //if(primaryService == nil || [primaryService isPrimary] == false) {
<   if(primaryService == nil) {
---
>   if(primaryService == nil || [primaryService isPrimary] == false) {
417c415
<     NSLog(@"Found service: %@, isPrimary %d", [s.UUID UUIDString], [s isPrimary]);
---
>     NSLog(@"Found service: %@", [s.UUID UUIDString]);
502,504c500
<
<   //if([descriptor.characteristic.service isPrimary]) {
<   if (true) {
---
>   if([descriptor.characteristic.service isPrimary]) {
513,524c509,510
<   //int value = [descriptor.value intValue];
<   //[result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
<   @try {
<     if  ([descriptor.value isKindOfClass:NSNumber.class]) {
<       int value = [descriptor.value intValue];
<       [result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
<     } else {
<       [result setValue:descriptor.value];
<     }
<   } @catch (NSException *exception) {
<     NSLog(@"Exception %@", exception.reason);
<   }
---
>   int value = [descriptor.value intValue];
>   [result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
542,543c528
<   //if([descriptor.characteristic.service isPrimary]) {
<   if (true) {
---
>   if([descriptor.characteristic.service isPrimary]) {
679,680c664
<   //[result setIsPrimary:[service isPrimary]];
<   [result setIsPrimary:true];
---
>   [result setIsPrimary:[service isPrimary]];
682,683d665
<   NSLog(@"service isPrimary:%d", [service isPrimary]);
<
713,715c695
<   //if([characteristic.service isPrimary]) {
<   if (true) {
<     NSLog(@"Primary");
---
>   if([characteristic.service isPrimary]) {
732,743c712,713
<   //int value = [descriptor.value intValue];
<   //[result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
<   @try {
<     if  ([descriptor.value isKindOfClass:NSNumber.class]) {
<       int value = [descriptor.value intValue];
<       [result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
<     } else {
<       [result setValue:descriptor.value];
<     }
<   } @catch (NSException *exception) {
<     NSLog(@"Exception %@", exception.reason);
<   }
---
>   int value = [descriptor.value intValue];
>   [result setValue:[NSData dataWithBytes:&value length:sizeof(value)]];
```

closed 2023-05-14 by endrelovas