pauldemarco/flutter_blue #658: convert BluetoothDevice to json

Akshay-04, 2020-08-16:
I am implementing an app where i need to send to send bluetoothdevice back to firebase (or local storage)and retrieve it from there and again use it.I could not find a solution where I could encode bluetoothdevice and then decode it as bluetoothdevice as we have no constructor to call.

Mirageofmage, 2020-08-18:
In the documentation there's a constructor for creating a bluetooth device from a [JSON constructor](https://pub.dev/documentation/flutter_blue/latest/gen_flutterblue.pb/BluetoothDevice/BluetoothDevice.fromJson.html)

Akshay-04, 2020-08-18:
Yes! I did find it but coundnt find a way to convert to json format.The dart json encode method does not work and toJson method is not implemented in this package.I looked at other package which is flutter_bluetooth which does include tojson method but the package itself doesn't look useful for my usecase

Mirageofmage, 2020-08-18:
The [BluetoothDevice](https://pub.dev/documentation/bluetooth/latest/gen_flutterblue.pb/BluetoothDevice-class.html) class has an inherited function named writeToJson(), which returns a JSON string with the device encoded. I haven't tested the functionality myself but this looks like the right direction to go.