pauldemarco/flutter_blue #662: I/flutter (13169): Error starting scan.

Amin-Matola, 2020-08-25:
Hello there, I added **flutter_blue** inside *pubspec.yaml* as:

```dependencies:
  flutter:
    sdk: flutter
  flutter_blue: #0.6.3+1
    git:
      url: git://github.com/pauldemarco/flutter_blue.git```

____________________
and added permissions inside **android/app/src/main/AndroidManifest.xml** as:
____________________
```<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.amin.bluewatch">
    <!-- io.flutter.app.FlutterApplication is an android.app.Application that
         calls FlutterMain.startInitialization(this); in its onCreate method.
         In most cases you can leave this as-is, but you if you want to provide
         additional functionality it is fine to subclass or reimplement
         FlutterApplication and put your custom class here. -->
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <application...```

_______________________
and in ios/Runner/info.plist as:
_______________________

```<dict>
    <key>NSBluetoothAlwaysUsageDescription</key>
    <string>Need BLE permission</string>
    <key>NSBluetoothPeripheralUsageDescription</key>
    <string>Need BLE permission</string>
    <key>NSLocationAlwaysAndWhenInUseUsageDescription</key>
    <string>Need Location permission</string>
    <key>NSLocationAlwaysUsageDescription</key>
    <string>Need Location permission</string>
    <key>NSLocationWhenInUseUsageDescription</key>
    <string>Need Location permission</string>

	<key>CFBundleDevelopmentRegion</key>...```
___________________________
and my scan test code went as:
___________________________

```floatingActionButton: FloatingActionButton(
        onPressed: () {

          FlutterBlue flutterBlue = FlutterBlue.instance;

          // Start scanning
          flutterBlue.startScan(timeout: Duration(seconds: 4));

          // Listen to scan results
           var subscription = flutterBlue.scanResults.listen((results) {
              
                // do something with scan results
                for (ScanResult r in results) {
                       
                       print('${r.device.name} found! rssi: ${r.rssi}');
                      
               }
          });

          // Stop scanning
          flutterBlue.stopScan();
        },
        child: Icon(Icons.search),
)```

______________________________
and below is the results I got inside the console...
______________________________

```D/BluetoothAdapter(23833): stopLeScan()
D/BluetoothAdapter(23833): startLeScan(): [Ljava.util.UUID;@41fba538
I/flutter (23833): Error starting scan.
E/flutter (23833): [ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: PlatformException(startScan, getBluetoothLeScanner() is null. Is the Adapter on?, java.lang.IllegalStateException: getBluetoothLeScanner() is null. Is the Adapter on?
E/flutter (23833): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.startScan18(FlutterBluePlugin.java:850)
E/flutter (23833): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.startScan(FlutterBluePlugin.java:751)
E/flutter (23833): 	at com.pauldemarco.flutter_blue.FlutterBluePlugin.onMethodCall(FlutterBluePlugin.java:255)
E/flutter (23833): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:230)
E/flutter (23833): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
E/flutter (23833): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:692)
E/flutter (23833): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (23833): 	at android.os.MessageQueue.next(MessageQueue.java:138)
E/flutter (23833): 	at android.os.Looper.loop(Looper.java:123)
E/flutter (23833): 	at android.app.ActivityThread.main(ActivityThread.java:5584)
E/flutter (23833): 	at java.lang.reflect.Method.invokeNative(Native Method)
E/flutter (23833): 	at java.lang.reflect.Method.invoke(Method.java:515)
E/flutter (23833): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1268)
E/flutter (23833): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1084)
E/flutter (23833): 	at dalvik.system.NativeStart.main(Native Method)
E/flutter (23833): )
E/flutter (23833): #0      FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:124:7)
E/flutter (23833): <asynchronous suspension>
E/flutter (23833): #1      FlutterBlue.startScan (package:flutter_blue/src/flutter_blue.dart:162:11)
E/flutter (23833): #2      _BlueWatchState.build.<anonymous closure> (package:bluewatch/main.dart:95:23)
E/flutter (23833): #3      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:992:19)
E/flutter (23833): #4      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:1098:38)
E/flutter (23833): #5      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:184:24)
E/flutter (23833): #6      TapGestureRecognizer.handleTapUp (package:flutter/src/gestures/tap.dart:524:11)
E/flutter (23833): #7      BaseTapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:284:5)
E/flutter (23833): #8      BaseTapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:219:7)
E/flutter (23833): #9      PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:477:9)
E/flutter (23833): #10     PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:78:12)
E/flutter (23833): #11     PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:124:9)
E/flutter (23833): #12     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:377:8)
E/flutter (23833): #13     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:122:18)
E/flutter (23833): #14     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:108:7)
E/flutter (23833): #15     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:220:19)
E/flutter (23833): #16     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:200:22)
E/flutter (23833): #17     GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:158:7)
E/flutter (23833): #18     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:104:7)
E/flutter (23833): #19     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:88:7)
E/flutter (23833): #20     _rootRunUnary (dart:async/zone.dart:1206:13)
E/flutter (23833): #21     _CustomZone.runUnary (dart:async/zone.dart:1100:19)
E/flutter (23833): #22     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1005:7)
E/flutter (23833): #23     _invoke1 (dart:ui/hooks.dart:267:10)
E/flutter (23833): #24     _dispatchPointerDataPacket (dart:ui/hooks.dart:176:5)
E/flutter (23833): 
```
____________________________________
I wonder what I am messing, or missing here to have it not working on my side?

javiervalero31, 2020-09-04:
Hello, I Have the same issue !

ugifractal, 2020-11-14:
I got similar issue.
```
E/flutter ( 5034): [ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: PlatformException(error, Field devices_ for b.c.a.u not found. Known fields are [private b.b.a.b0$i b.c.a.u.e, private static final b.c.a.u b.c.a.u.f, private static volatile b.b.a.a1 b.c.a.u.g], null, java.lang.RuntimeException: Field devices_ for b.c.a.u not found. Known fields are [private b.b.a.b0$i b.c.a.u.e, private static final b.c.a.u b.c.a.u.f, private static volatile b.b.a.a1 b.c.a.u.g]
E/flutter ( 5034): 	at b.b.a.v0.a(Unknown Source:72)
E/flutter ( 5034): 	at b.b.a.v0.a(Unknown Source:715)
E/flutter ( 5034): 	at b.b.a.v0.a(Unknown Source:12)
E/flutter ( 5034): 	at b.b.a.k0.a(Unknown Source:60)
E/flutter ( 5034): 	at b.b.a.k0.a(Unknown Source:49)
E/flutter ( 5034): 	at b.b.a.d1.a(Unknown Source:17)
E/flutter ( 5034): 	at b.b.a.d1.a(Unknown Source:4)
E/flutter ( 5034): 	at b.b.a.z.m(Unknown Source:4)
E/flutter ( 5034): 	at b.b.a.z$a.a(Unknown Source:9)
E/flutter ( 5034): 	at b.b.a.z$a.b(Unknown Source:0)
E/flutter ( 5034): 	at b.c.a.b.onMethodCall(Unknown Source:1336)
E/flutter ( 5034): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(Unknown Source:17)
E/flutter ( 5034): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(Unknown Source:57)
E/flutter ( 5034): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter ( 5034): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter ( 5034): 	at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter ( 5034): 	at android.os.Looper.loop(Looper.java:174)
E/flutter ( 5034): 	at android.app.ActivityThread.main(ActivityThread.java:7397)
E/flutter ( 5034): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 5034): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/flutter ( 5034): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
E/flutter ( 5034): )
E/flutter ( 5034): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:582)
E/flutter ( 5034): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:159)
E/flutter ( 5034): <asynchronous suspension>
E/flutter ( 5034): #2      FlutterBlue.connectedDevices.<anonymous closure> (package:flutter_blue/src/flutter_blue.dart:72)
E/flutter ( 5034): <asynchronous suspension>
E/flutter ( 5034): 
I/flutter ( 5034): Error starting scan.
E/flutter ( 5034): [ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: PlatformException(startScan, Field androidScanMode_ for b.c.a.s0 not found. Known fields are [private int b.c.a.s0.e, private b.b.a.b0$i b.c.a.s0.f, private boolean b.c.a.s0.g, private static final b.c.a.s0 b.c.a.s0.h, private static volatile b.b.a.a1 b.c.a.s0.i], java.lang.RuntimeException: Field androidScanMode_ for b.c.a.s0 not found. Known fields are [private int b.c.a.s0.e, private b.b.a.b0$i b.c.a.s0.f, private boolean b.c.a.s0.g, private static final b.c.a.s0 b.c.a.s0.h, private static volatile b.b.a.a1 b.c.a.s0.i]
E/flutter ( 5034): 	at b.b.a.v0.a(Unknown Source:72)
E/flutter ( 5034): 	at b.b.a.v0.a(Unknown Source:715)
E/flutter ( 5034): 	at b.b.a.v0.a(Unknown Source:12)
E/flutter ( 5034): 	at b.b.a.k0.a(Unknown Source:60)
E/flutter ( 5034): 	at b.b.a.k0.a(Unknown Source:49)
E/flutter ( 5034): 	at b.b.a.d1.a(Unknown Source:17)
E/flutter ( 5034): 	at b.b.a.d1.a(Unknown Source:4)
E/flutter ( 5034): 	at b.b.a.z$a.a(Unknown Source:9)
E/flutter ( 5034): 	at b.b.a.z$a.a(Unknown Source:4)
E/flutter ( 5034): 	at b.b.a.z$a.a(Unknown Source:0)
E/flutter ( 5034): 	at b.b.a.a$a.a(Unknown Source:2)
E/flutter ( 5034): 	at b.c.a.b.a(Unknown Source:10)
E/flutter ( 5034): 	at b.c.a.b.onMethodCall(Unknown Source:1414)
E/flutter ( 5034): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(Unknown Source:17)
E/flutter ( 5034): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(Unknown Source:57)
E/flutter ( 5034): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter ( 5034): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter ( 5034): 	at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter ( 5034): 	at android.os.Looper.loop(Looper.java:174)
E/flutter ( 5034): 	at android.app.ActivityThread.main(ActivityThread.java:7397)
E/flutter ( 5034): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 5034): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/flutter ( 5034): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)
E/flutter ( 5034): , null)
E/flutter ( 5034): #0      FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:124)
E/flutter ( 5034): <asynchronous suspension>
E/flutter ( 5034): <asynchronous suspension>
E/flutter ( 5034): 

```

spekary, 2020-11-20:
I have the same issue as above. However, the problem only happens when I install a release build of my application. If I install the debug build, it works fine:

I/FlutterBluePlugin(28742): setup
I/flutter (28742): Error starting scan.
E/flutter (28742): [ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: PlatformException(startScan, Field androidScanMode_ for b.b.a.s0 not found. Known fields are [private int b.b.a.s0.e, private b.a.a.b0$i b.b.a.s0.f, private boolean b.b.a.s0.g, private static final b.b.a.s0 b.b.a.s0.h, private static volatile b.a.a.a1 b.b.a.s0.i], java.lang.RuntimeException: Field androidScanMode_ for b.b.a.s0 not found. Known fields are [private int b.b.a.s0.e, private b.a.a.b0$i b.b.a.s0.f, private boolean b.b.a.s0.g, private static final b.b.a.s0 b.b.a.s0.h, private static volatile b.a.a.a1 b.b.a.s0.i]
E/flutter (28742): 	at b.a.a.v0.a(Unknown Source:72)
E/flutter (28742): 	at b.a.a.v0.a(Unknown Source:715)
E/flutter (28742): 	at b.a.a.v0.a(Unknown Source:12)
E/flutter (28742): 	at b.a.a.k0.a(Unknown Source:60)
E/flutter (28742): 	at b.a.a.k0.a(Unknown Source:49)
E/flutter (28742): 	at b.a.a.d1.a(Unknown Source:17)
E/flutter (28742): 	at b.a.a.d1.a(Unknown Source:4)
E/flutter (28742): 	at b.a.a.z$a.a(Unknown Source:9)
E/flutter (28742): 	at b.a.a.z$a.a(Unknown Source:4)
E/flutter (28742): 	at b.a.a.z$a.a(Unknown Source:0)
E/flutter (28742): 	at b.a.a.a$a.a(Unknown Source:2)
E/flutter (28742): 	at b.b.a.b.b(Unknown Source:10)
E/flutter (28742): 	at b.b.a.b.a(Unknown Source:1414)
E/flutter (28742): 	at c.a.c.a.j$a.a(Unknown Source:17)
E/flutter (28742): 	at io.flutter.embedding.engine.e.b.a(Unknown Source:57)
E/flutter (28742): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter (28742): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter (28742): 	at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter (28742): 	at android.os.Looper.loop(Looper.java:174)
E/flutter (28742): 	at android.app.ActivityThread.main(ActivityThread.java:7386)
E/flutter (28742): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (28742): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
E/flutter (28742): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
E/flutter (28742): , null)
E/flutter (28742): #0      FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:124)
E/flutter (28742): <asynchronous suspension>
E/flutter (28742): <asynchronous suspension>
E/flutter (28742):

ugifractal, 2020-11-20:
I fixed this issue on release build by disable shrinkResources at ```android/app/build.gradle```

```
buildTypes {
        release {
            // TODO: Add your own signing config for the release build.
            // Signing with the debug keys for now, so `flutter run --release` works.
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
            //proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
```

spekary, 2020-11-20:
Neither 

shrinkResources false

Nor 

minifyEnabled false

worked for me.

This is on Android 8.1.0.

gksrb4, 2020-11-20:
`flutter build apk --no-shrink`
worked for me.

dhaval-sabhaya, 2020-12-09:
Hello flutter_blue team,

igorczapski, 2020-12-17:
I just completely excluded flutter blue classes in proguard rules.

-keep class com.pauldemarco.flutter_blue.** { *; }

MrCsabaToth, 2020-12-28:
I was skeptic but the ProGuard rule helped!

nicolosponziello, 2021-07-01:
are there any updates on this issue? disable shrinking solves it but is unconvenient

MrCsabaToth, 2021-07-01:
> are there any updates on this issue? disable shrinking solves it but is unconvenient

Add the ProGuard rules (creating the file if you don't have one already). Minimum these rules:
```
-keep class com.pauldemarco.flutter_blue.** { *; }
-keepclassmembernames class com.pauldemarco.flutter_blue.* { *; }
```
This is not a Flutter Blue specific problem.
For more rules see this blog post: https://csaba.page/blog/flutter-android-crash-remedies.html

aygunbi, 2021-11-17:
> Merhaba flutter_blue ekibi, Biz de aynı sorunla karşı karşıyayız.
> 
> Sorun yalnızca uygulamamın bir sürüm derlemesini yüklediğimde oluyor. Hata ayıklama derlemesini yüklersem düzgün çalışır:
> 
> ```
> buildTypes {
>         release {
>             signingConfig signingConfigs.debug
>             shrinkResources false
>             minifyEnabled false
>         }
>     }
> ```
> 
> Android/app/build.gradle konumunda küçültme kaynakları ve minifyEnabled devre dışı bırakılarak sürüm derlemesinde bu sorun düzeltildi
> 
> Ama ProGuard'ı devre dışı bıraktığı için garip. Peki ya android apk dosya kaynağı menkul kıymetleri.
> 
> Bu yüzden ProGuard etkinleştirmesi ile çalışması için aynısı için proaktif bir çözüme ihtiyacı vardır.
> 
> Bu konuya verdiğiniz yanıt için çok teşekkür ederim Teşekkürler!

👍 worked, thanks 👍👍👍👍👍👍

supriyavani, 2022-02-04:
> Add the ProGuard rules (creating the file if you don't have one already). Minimum these rules:
>
> ```
> -keep class com.pauldemarco.flutter_blue.** { *; }
> -keepclassmembernames class [...]

supriyavani, 2022-02-04:
hi there i tried all the above added proguard,disabled  shrinkResources false
            minifyEnabled false and advices from others too still the issue exists my bluetooth app workd fine on the emulator but doesnt work on real device

supriyavani, 2022-02-04:
somebody pleaseeeeeeeeeeeeeeee help me save my life i am stuck here

boskokg, 2022-02-04:
@supriyavani this plugin is dead. Further work is done on flutter_blue_plus.

afranmuzammil, 2022-03-01:
@boskokg I am getting the same problem with flutter_blue_plus

MuksithMRA, 2022-08-27:
> Add the ProGuard rules (creating the file if you don't have one already). Minimum these rules:
>
> ```
> -keep class com.pauldemarco.flutter_blue.** { *; }
> -keepclassmembernames class [...]

> Add the ProGuard rules (creating the file if you don't have one already). Minimum these rules:
>
> ```
> -keep class com.pauldemarco.flutter_blue.** { *; }
> -keepclassmembernames class [...]

It Worked Bro. Thank you

parzibyte, 2022-10-08:
Same issue on october 2022: Unhandled Exception: PlatformException(device_state_error, Field remoteId_ for k.n not found...
What worked for me was adding `shrinkResources false` and `minifyEnabled false` on `android/app/build.gradle`.
```
buildTypes {
        release {
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
        }
    }
```
I am using flutter_blue: ^0.8.0
`flutter build apk --no-shrink` did not work, neither did modifying proguard files.

lazarekh, 2023-12-16:
I have the same issue on flutter_blue_plus. I can do startScan(), but get no device. Although, my device own bluetooth can scan and find devices nearby. I experience it in debug mode.