pauldemarco/flutter_blue #664: Could not download protoc-gen-javalite-3.0.0-osx-x86_64.exe

guominfang, 2020-08-26:
```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue:generateDebugProto'.
> Could not resolve all files for configuration ':flutter_blue:protobufToolsLocator_javalite'.
   > Could not download protoc-gen-javalite-3.0.0-osx-x86_64.exe (com.google.protobuf:protoc-gen-javalite:3.0.0)
      > Could not get resource 'https://jcenter.bintray.com/com/google/protobuf/protoc-gen-javalite/3.0.0/protoc-gen-javalite-3.0.0-osx-x86_64.exe'.
         > Could not GET 'https://jcenter.bintray.com/com/google/protobuf/protoc-gen-javalite/3.0.0/protoc-gen-javalite-3.0.0-osx-x86_64.exe'.
            > Read timed out

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 37s
Exception: Gradle task assembleDebug failed with exit code 1

```

```
[✓] Flutter (Channel stable, 1.20.2, on Mac OS X 10.15.6 19G2021, locale
    zh-Hans-CN)
    • Flutter version 1.20.2 at /Users/mac/Library/Flutter/flutter
    • Framework revision bbfbf1770c (13 days ago), 2020-08-13 08:33:09 -0700
    • Engine revision 9d5b21729f
    • Dart version 2.9.1
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.2)
    • Android SDK at /Users/mac/Library/Android/sdk
    • Platform android-30, build-tools 30.0.2
    • ANDROID_HOME = /Users/mac/Library/Android/sdk
    • Java binary at: /Applications/Android
      Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build
      1.8.0_242-release-1644-b3-6222593)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.6)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.6, Build version 11E708
    • CocoaPods version 1.9.3

[✓] Android Studio (version 4.0)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 48.1.2
    • Dart plugin version 193.7361
    • Java version OpenJDK Runtime Environment (build
      1.8.0_242-release-1644-b3-6222593)

[✓] VS Code (version 1.48.0)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.13.2

[✓] Connected device (1 available)
    • HLK AL10 (mobile) • DWVNW19730006016 • android-arm64 • Android 9 (API 28)

• No issues found!

```

guominfang, 2020-08-26:
Oh, it's my network problem

closed 2020-08-26 by guominfang

LeoAiolia, 2022-03-28:
same issue, how to fix it ?   @guominfang

LeoAiolia, 2022-03-28:
my mac is m1 mbp

guominfang, 2022-04-01:
@LeoAiolia 我是网络问题，有墙