pauldemarco/flutter_blue #665: Android scan after pair

xelons, 2020-08-28:
Hi,
I have a weird problem with Android devices.
If my ble module is not paired (bonded) with Android, then it is listed after doing a flutter_blue scan, once it is paired with a pin, it does not show again, even after a restart.

To be able to see it on scan results, I have to "forget" unpair the device, then it shows again in the scan list.

On iOS it works fine, even after I pair the device, it still show on the scan list.
Any help is appreciated!

javiervalero31, 2020-09-04:
Hello @xelons, Can You comment me how did you do scan the nearby devices and how did you listen it in the through  FlutterBlue.instance.scan Results ???

xelons, 2020-09-05:
@javiervalero31 I used the exact example code, once I connect to a device which as a static pin, I get the popup to enter the pin, then once paired successfully I no longer see the device in the scan result list nor on the connected devices list.

When I go to Bluetooth devices under Android menu, and manually forget(unpair) the device, then I can see it again on scan.

Works fine on iOS.

rezendegc, 2020-09-09:
Up. After I connect with device.connect(), the device shows on FlutterBlue.instance.connectedDevices, but after I try to write something the OS asks for Pin code and the device disappear from the connectedDevices list.

xelons, 2020-09-11:
This issue started happening on IOS as well... Once I pair with a pin and the device is bonded, I can no longer connect to it, I have to unpair/forget the device to be able to connect to it again.