pauldemarco/flutter_blue #666: Can I use flutter blue to act as a BLE server?

Ajorona, 2020-09-01:
From the docs I am not sure if this is supported. I would like to create a server and client stream in order to support a Custom Serial Chat Service:

[image]

This is a custom protocol on the ATSAMB11 board manufactured by MicroChip Technologies.

orevial, 2020-09-14:
> I'm not sure iOS supports BLE in peripheral mode

Apparently iOS supports BLE in peripheral mode, see [Core Bluetooth Programming Guide](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/PerformingCommonPeripheralRoleTasks/PerformingCommonPeripheralRoleTasks.html#//apple_ref/doc/uid/TP40013257-CH4-SW1) for more information

Ajorona, 2020-09-16:
I believe the answer is no for this plugin.

closed 2020-09-16 by Ajorona