pauldemarco/flutter_blue #670: flutter_blue does not work on IOS device

decolon, 2020-09-05:
This seems to be related to #406, but starting a new thread since the old one is not very active. 

When I build with Android, I have no problems. When I build without flutter_blue in my pubspec, no problems. As soon as I add flutter_blue in my pubspec, and build for my iPhone, I get this error
```
Running "flutter pub get" in nobobo_pantry...
Launching lib/main.dart on iPhone (4) in debug mode...
Automatically signing iOS for device deployment using specified development team in Xcode project: 8R4KTRC86F
Running pod install...
Error: CocoaPods's specs repository is too out-of-date to satisfy dependencies.
To update the CocoaPods specs, run:
  pod repo update

Exception: Error running pod install
CocoaPods' output:
↳
  Preparing

Analyzing dependencies

Inspecting targets to integrate
  Using `ARCHS` setting to build architectures of target `Pods-Runner`: (``)

Finding Podfile changes
  A flutter_blue
  - Flutter
  - camera
  - cloud_firestore
  - firebase_auth
  - firebase_core
  - firebase_ml_vision
  - firebase_storage
  - flutter_exif_rotation
  - path_provider
  - vibration

Fetching external sources
-> Fetching podspec for `Flutter` from `Flutter`
-> Fetching podspec for `camera` from `.symlinks/plugins/camera/ios`
-> Fetching podspec for `cloud_firestore` from `.symlinks/plugins/cloud_firestore/ios`
-> Fetching podspec for `firebase_auth` from `.symlinks/plugins/firebase_auth/ios`
-> Fetching podspec for `firebase_core` from `.symlinks/plugins/firebase_core/ios`
-> Fetching podspec for `firebase_ml_vision` from `.symlinks/plugins/firebase_ml_vision/ios`
-> Fetching podspec for `firebase_storage` from `.symlinks/plugins/firebase_storage/ios`
-> Fetching podspec for `flutter_blue` from `.symlinks/plugins/flutter_blue/ios`
-> Fetching podspec for `flutter_exif_rotation` from `.symlinks/plugins/flutter_exif_rotation/ios`
-> Fetching podspec for `path_provider` from `.symlinks/plugins/path_provider/ios`
-> Fetching podspec for `vibration` from `.symlinks/plugins/vibration/ios`

Resolving dependencies of `Podfile`
  CDN: trunk Relative path: CocoaPods-version.yml exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: all_pods_versions_0_3_5.txt exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: all_pods_versions_6_3_6.txt exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/6/3/6/FirebaseAuth/6.8.0/FirebaseAuth.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: all_pods_versions_e_c_d.txt exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/e/c/d/Protobuf/4.0.0-rc1/Protobuf.podspec.json exists! Returning local because checking is only perfomed in repo update
[!] CocoaPods could not find compatible versions for pod "Protobuf":
  In snapshot (Podfile.lock):
    Protobuf (= 3.13.0, ~> 3.12)

  In Podfile:
    flutter_blue (from `.symlinks/plugins/flutter_blue/ios`) was resolved to 0.0.1, which depends on
      flutter_blue/Protos (= 0.0.1) was resolved to 0.0.1, which depends on
        Protobuf (~> 3.11.4)
[... 61 lines cut ...]
    • Platform android-30, build-tools 30.0.0
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.6)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.6, Build version 11E708
    • CocoaPods version 1.9.3

[✓] Android Studio (version 4.0)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 48.1.2
    • Dart plugin version 193.7361
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)

[✓] VS Code (version 1.46.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.12.0

[✓] Connected device (1 available)
    • iPhone (4) (mobile) • 5ae9753c41cae0609366ed62da85ce52106a6082 • ios • iOS 13.6

• No issues found!
```

LoveCcheng, 2021-04-01:
I have the same situation and it hasn't been resolved yet. Have you solved it yet?

ju851zel, 2021-09-15:
Have you tried adding the package vie ```flutter_blue: ^0.8.0``` in the pubspec.yaml and not the git repo directly?

RodrigoLScorza, 2021-11-04:
> Have you tried adding the package vie `flutter_blue: ^0.8.0` in the pubspec.yaml and not the git repo directly?

Yes, I did but have same problem

devSumanrazz, 2022-03-29:
@pauldemarco Any fix on this?