pauldemarco/flutter_blue #672: Flutter Blue broadcast message read for authentication.

saqib556612, 2020-09-05:
I am sending a broadcast message from Esp32 when i connect to the Esp32 from flutter application using flutter blue i can't read the broadcast message ? is this feature implemented ?

RedsAnDev, 2020-09-15:
Hi, I'm working on a GATT Profile and using Flutter_Blue I had no specific issue.  How can you read a single message in a single scan? I'm not able to reproduce, can you share your code? Of course, you can omit the business logic parts.

saqib556612, 2020-10-05:
actually when i read manufacturer data it gives me wrong packet . i am sending specific packet from BLE device and sometime it can read the packet and sometime it can't and get the wrong packet.