pauldemarco/flutter_blue #674: Asynchronous call to discover services does not return on connected device.

anthonyorona, 2020-09-09:
I make multiple async calls and await each in turn (because they must be done in sequence for this protocol).

The call to await device.discoverServices() never returns.

     /
      Future<void> connectMyProtocol(BluetoothDevice device) async {
          device = device;
          await device.connect(timeout: Duration(seconds: 5));
          await device.requestMtu(250);
          List<BluetoothService> services = await device.discoverServices();
         // CODE DOES NOT  

          myService = services.firstWhere((service) => service.uuid.toString() == MyServiceUUID);

          BluetoothCharacteristic endpointCharacteristic = scService.characteristics.firstWhere(
                (c) => c.uuid.toString() == SCSEndpointCharacteristic
          );
          await endpointCharacteristic.setNotifyValue(true);

          endpointCharacteristic.value.listen((List<int> value) {
               return;
           }
          _serverInitialized = true;
      }
`

LOGS:
I/flutter (18994): BEFORE CONNECT
D/BluetoothGatt(18994): connect() - device: AB:89:67:45:11:FF, auto: true
D/BluetoothGatt(18994): registerApp()
D/BluetoothGatt(18994): registerApp() - UUID=74d1b084-5572-4ce2-9fe3-b5eafbec4105
D/BluetoothGatt(18994): onClientRegistered() - status=0 clientIf=10
D/BluetoothGattServer(18994): onServerConnectionState() - status=0 serverIf=9 device=AB:89:67:45:11:FF
D/BluetoothGatt(18994): onClientConnectionState() - status=0 clientIf=10 device=AB:89:67:45:11:FF
D/FlutterBluePlugin(18994): [onConnectionStateChange] status: 0 newState: 2
D/FlutterBluePlugin(18994): mDevices size: 1
D/BluetoothGatt(18994): configureMTU() - device: AB:89:67:45:11:FF mtu: 250
I/flutter (18994): BEFORE DISCOVER SERVICES
D/BluetoothGatt(18994): discoverServices() - device: AB:89:67:45:11:FF
D/BluetoothGattServer(18994): onConnectionUpdated() - Device=AB:89:67:45:11:FF interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(18994): onConnectionUpdated() - Device=AB:89:67:45:11:FF interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(18994): onConfigureMTU() - Device=AB:89:67:45:11:FF mtu=250 status=0
D/FlutterBluePlugin(18994): [onMtuChanged] mtu: 250 status: 0
D/BluetoothGattServer(18994): onConnectionUpdated() - Device=AB:89:67:45:11:FF interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(18994): onConnectionUpdated() - Device=AB:89:67:45:11:FF interval=36 latency=0 timeout=500 status=0
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1
D/FlutterBluePlugin(18994): mDevices size: 1

goodxianping, 2020-09-11:
the same problem on my phone.sadly!

RedsAnDev, 2020-09-15:
I have same problem and I will share some fix that I use:

- **firstWhere** need clause orElse: ()=> null. Of course you need to add a little logic for null value
- I prefer to set listen on stream BluetoothChr before to setNotifyValue(True).

I hope this can help you.

corecode, 2020-11-22:
You need to wait until the MTU negotiation is done.

I tried this and it works, but likely it is not robust:
``` dart
    await device.requestMtu(512);
    await device.mtu.skip(1).first;
```

akash-flutterdev, 2021-03-18:
i am facing same issue any one can tell me how to fix it please

akash-flutterdev, 2021-03-18:
> I have same problem and I will share some fix that I use:
> 
> * **firstWhere** need clause orElse: ()=> null. Of course you need to add a little logic for null value
> * I prefer to set listen on stream BluetoothChr before to setNotifyValue(True).
> 
> I hope this can help you.

hii can you tell me how to fix this ?