pauldemarco/flutter_blue #681: Connect taking too long

HugoSart, 2020-09-15:
After scanning some devices and calling the connect method, it takes more than 1 minute sometimes, and sometimes it never connects. Also, something that could be related, the state stream doesn't look correct: it's either "connected" or "disconnected". I'm never able to see the "connecting" or "disconnecting" state.

Sample repository: https://github.com/HugoSart/flutter_blue_hangs

arturo393, 2020-11-12:
Same situation for me.
Sometimes it connects in 5 secs and sometimes it connects in more than 5 attempts. 
this is my connect method
```
  Future<void> connectTimeout(ScanResult result) async {
    await result.device
        .connect(autoConnect: true)
        .timeout(Duration(seconds: 5), onTimeout: () {
      log('${result.device.name} connecting timeout');
      result.device.disconnect();
        }).then((data) async {
          //TODO no lo puedo hacer acá porque se repite mucho
      await for (BluetoothDeviceState r in result.device.state) {
        log('${result.device.name} state: ${r}');
        if (r == BluetoothDeviceState.connected) {
          checkConnectedImei(result.device);
        }
      }

    }).catchError((e) {
      log(' ${result.device.name}  connected state: ${result.device.state} $e');
    }).whenComplete(()  {});
  }
```

jyuno426, 2021-02-03:
In some cases, you can solve it just by setting [autoConnect: false]. I don't know why it works, but it works pretty well. 

https://github.com/pauldemarco/flutter_blue/issues/106#issuecomment-467697537

fyredave, 2021-06-23:
> 
> 
> In some cases, you can solve it just by setting [autoConnect: false]. I don't know why it works, but it works pretty well.
> 
> [#106 (comment)](https://github.com/pauldemarco/flutter_blue/issues/106#issuecomment-467697537)

Setting autoConnect to false when the Platform was Android resolved my issue for a specific BLE device that would connect with iOS but not Android