pauldemarco/flutter_blue #685: Cannot catch timeouterror when device connect timeout

xiang23, 2020-09-22:
my test code
```
 try {
      await device.connect(timeout: Duration(seconds: 0), autoConnect: false);
    } catch (e) {
      debugPrint('catch $e');
    }
```
and next you can see
```
E/flutter (26345): [ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: TimeoutException after 0:00:00.000000: Failed to connect in time.
E/flutter (26345): #0      BluetoothDevice.connect.<anonymous closure> (package:flutter_blue/src/bluetooth_device.dart:33:9)
E/flutter (26345): #1      _rootRun (dart:async/zone.dart:1182:47)
E/flutter (26345): #2      _CustomZone.run (dart:async/zone.dart:1093:19)
E/flutter (26345): #3      _CustomZone.runGuarded (dart:async/zone.dart:997:7)
E/flutter (26345): #4      _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
E/flutter (26345): #5      _rootRun (dart:async/zone.dart:1190:13)
E/flutter (26345): #6      _CustomZone.run (dart:async/zone.dart:1093:19)
E/flutter (26345): #7      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:1021:23)
E/flutter (26345): #8      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
E/flutter (26345): #9      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:397:19)
E/flutter (26345): #10     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:428:5)
E/flutter (26345): #11     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)
```
if i want catch error maybe only like this:
```
 runZoned(()async{
      await device.connect(timeout: Duration(seconds: 0), autoConnect: false);
    },onError: (){
      debugPrint('catch $e');
    });
```

Mirageofmage, 2020-10-06:
You have a timeout set to 0 seconds, odds are the connection times out instantly before the device has a chance to connect, try either extending the timer or removing it entirely

luckbuff, 2020-11-26:
I set timeout to 10s，but I cant catch the exception.

Chakib-Temal, 2020-11-26:
Solved  : https://github.com/pauldemarco/flutter_blue/issues/299#issuecomment-521203753

closed 2020-11-26 by xiang23