pauldemarco/flutter_blue #687: I am trying to read data from blood pressure monitor but unable to get actual value of SYS , HEART BEAT,  etc .

prkay, 2020-09-24:
I am trying to connect the blood pressure monitor using flutter blue.
I get characteristics value 
[254, 32, 39, 0, 117, 0, 85, 0, 95, 0, 228, 7, 9, 24, 16, 9, 58, 105, 0, 1]
[117, 85, 95, 228, 1801, 6160, 2362, 26880, 256, 26938, 2320, 6153, 2020, 95, 85, 117, 39] 

Please provide details.

BananaNeil, 2021-02-11:
Can you provide your code sample on the flutter side?

What exactly are you expecting from your BLE device?

akashkarli, 2021-02-16:
> I am trying to connect the blood pressure monitor using flutter blue.
> I get characteristics value
> [254, 32, 39, 0, 117, 0, 85, 0, 95, 0, 228, 7, 9, 24, 16, 9, 58, 105, 0, 1]
> [117, 85, 95, 228, 1801, 6160, 2362, 26880, 256, 26938, 2320, 6153, 2020, 95, 85, 117, 39]
> 
> Please provide details.

i think you have to decode in byte value in to bit and some having some calculation

prkay, 2021-02-22:
Thanks @BananaNeil @akashkarli 
But I got the solutions.  
value = [254, 32, 39, 0, 117, 0, 85, 0, 95, 0, 228, 7, 9, 24, 16, 9, 58, 105, 0, 1]
 systolicPressure: value[1]
 diastolicPressure: value[3]
heartRate: value[14]
9 to 12th index is for time of measurment and we have to get data for ServiceUuid = "00001810-0000-1000-8000-00805f9b34fb" && CharactersticUuid = "00002a35-0000-1000-8000-00805f9b34fb"

closed 2021-02-22 by prkay

amirr-dotcom, 2021-11-29:
@prkay  can you help me implement bp Machine With  Flutter

phan-dev, 2022-04-13:
> Thanks @BananaNeil @akashkarli But I got the solutions. value = [254, 32, 39, 0, 117, 0, 85, 0, 95, 0, 228, 7, 9, 24, 16, 9, 58, 105, 0, 1] systolicPressure: value[1] diastolicPressure: value[3] [...]

Hi @prkay 
How can you know that?
Could you please share the references or documents?

akashkarli, 2022-06-15:
> Please if any one can help me I need to implement blue tooth device

https://pub.dev/packages/flutter_blue/example
please refer this example and see how is it implement

fitterfly-parth, 2022-11-08:
I didn't get any value from below ids.

ServiceUuid = "00001810-0000-1000-8000-00805f9b34fb" && CharactersticUuid = "00002a35-0000-1000-8000-00805f9b34fb"

Can you help me to find value?

I'm using omron 7156T model and device have 11 readings.

prkay, 2022-11-08:
@fitterfly-parth  please send complete data I will provide you how can you fetch that

fitterfly-parth, 2023-01-31:
> @fitterfly-parth please send complete data I will provide you how can you fetch that

00002a35-0000-1000-8000-00805f9b34fb -> setNotifyValue(true)
00002a49-0000-1000-8000-00805f9b34fb -> read()

but not able to get readings. Can you please give me the step one by one how to fetch data from OMRON 7156T

smdevSi, 2024-02-22:
> Hi @prkay How can you know that? Could you please share the references or documents?

I would also like to know where to find this documentation