pauldemarco/flutter_blue #689: Connection Xiaomi MiBand with Dart Language (Flutter)

claytonsands, 2020-09-25:
Good afternoon, I need to integrate the xiaomi smart bracelet (miband 3) in the Dart language (Flutter), but I am having difficulty, there is very little support material regarding this integration specifically.

I found only one repository on GitHub that makes this call, I was able to connect via bluetooth with the bracelet but I can't receive and send alert data for example. The services are listed, but when I click on read or write nothing happens. Can anyone help me with this?

I used the plugin flutter_blue and the code of the example is here: https://github.com/pauldemarco/flutter_blue/blob/master/example/lib/main.dart

Dema323, 2020-10-04:
I was able to read some service codes. But no meaningfull data yet. Did you make any progress ?

maks, 2021-02-07:
I would suggest it's probably easier to use something like [NRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_AU&gl=US) to figure out how to use the services and characteristics on the device and then move onto using Flutter Blue to communicate with it.

Doing even the most basic research finds theres [already documentation regarding the need to do authentication with the miband over BLE](https://medium.com/@yogeshojha/i-hacked-xiaomi-miband-3-and-here-is-how-i-did-it-43d68c272391)
so I would suggest closing this issue @claytonsands unless there is something specific regarding flutter blue that does not work with that device.

Sherif-2001, 2022-12-25:
Have you finished this app??

Refaydeen, 2023-03-17:
hi guys , i  created a app for read a data from mi band ,i m getting the services but i cant able to read the heart and stepcount , is anyone worked on this can you help me

prineth, 2023-04-02:
Have anyone sort it out?

VishalVinayRam, 2023-12-30:
@Refaydeen could you tell me how did u buit the app where you getting data from the mi band .

kan15hka, 2024-03-31:
Hi guys any update on getting data from mi band