pauldemarco/flutter_blue #690: Bluetooth adapter is unknown

saqib556612, 2020-09-26:
i take an example code of flutter_blue package. sometimes when i open app its show me Bluetooth adapters is unknown even the bluetooth is on.
[image]

PhilColbert, 2020-12-23:
Also getting this issue.

Just on Iphone 10 so far.

Restarting the app solves this issue....

Any ideas ?

sercan-uzun, 2022-05-22:
I am using flutter_blue_plus. It is happening to me. I need reinstall, or re-run on flutter for working.

BonDizard, 2024-07-25:
Did the issue get resolved? For Android, I'm getting the following error:

[image]

We can see that Bluetooth is on, yet I'm getting the error. I just used the example code in the flutter_blue_plus package. For the device off state, I used the following code:

```
class BluetoothOffScreen extends StatelessWidget {
  final BluetoothAdapterState bluetoothAdapterState;

  const BluetoothOffScreen({super.key, required this.bluetoothAdapterState});

  @override
  Widget build(BuildContext context) {
    String message;
    IconData icon;
    Color iconColor;

    switch (bluetoothAdapterState) {
      case BluetoothAdapterState.unknown:
        message = 'Bluetooth Status Unknown';
        icon = Icons.help_outline;
        iconColor = Colors.amber;
        break;
      case BluetoothAdapterState.off:
        message = 'Bluetooth is Disabled';
        icon = Icons.bluetooth_disabled;
        iconColor = Colors.white;
        break;
      case BluetoothAdapterState.on:
        // This shouldn't reach this screen, you can navigate elsewhere.
        Navigator.of(context).pop(); // Remove from navigation stack
        return Container(); // Empty container to prevent rendering issues
      default:
        message = 'Unexpected State';
        icon = Icons.error_outline;
        iconColor = Colors.red;
    }

    return Scaffold(
      backgroundColor: Theme.of(context).colorScheme.error,
      body: Center(
        child: Column(
          mainAxisAlignment: MainAxisAlignment.center,
          crossAxisAlignment: CrossAxisAlignment.center,
          children: [
            Icon(
              icon,
              size: 72.0,
              color: iconColor,
            ),
            const Padding(padding: EdgeInsets.all(16.0)),
            Text(
              message,
              style: const TextStyle(
                fontFamily: 'Alatsi',
                color: Colors.white,
                fontSize: 30,
              ),
            ),
            const Padding(padding: EdgeInsets.all(8.0)),
            const Text(
              'Please enable Bluetooth to connect to devices.',
              style: TextStyle(
                fontFamily: 'Alatsi',
                color: ColorConstants.darkColor,
              ),
            ),
            const Padding(padding: EdgeInsets.all(16.0)),
            ReusableButton(
              onPressed: () async {
                await requestBluetoothPermission();
              },
              buttonText: 'Turn on Bluetooth',
            ),
          ],
        ),
      ),
    );
  }

  Future<void> requestBluetoothPermission() async {
    try {
      if (Platform.isAndroid) {
        await FlutterBluePlus.turnOn();
      }
    } catch (e) {
      CustomToast.showToast("Error Turning On: $e");
    }
  }
}
```

The below code is my main.dart. It's exactly as the example code, but still, I'm sending it if it helps:

```
void main() {
  FlutterBluePlus.setLogLevel(LogLevel.verbose, color: true);
  WidgetsFlutterBinding.ensureInitialized();
  runApp(
    const ProviderScope(
      child: FlutterBlueApp(),
    ),
  );
}

class FlutterBlueApp extends StatefulWidget {
  const FlutterBlueApp({Key? key}) : super(key: key);

  @override
  State<FlutterBlueApp> createState() => _FlutterBlueAppState();
}

class _FlutterBlueAppState extends State<FlutterBlueApp> {
  BluetoothAdapterState _adapterState = BluetoothAdapterState.unknown;

  late StreamSubscription<BluetoothAdapterState> _adapterStateStateSubscription;

  @override
  void initState() {
    super.initState();
    _adapterStateStateSubscription =
        FlutterBluePlus.adapterState.listen((state) {
      _adapterState = state;

      if (mounted) {
        setState(() {});
      }
      if (!mounted) return;
    });
  }

  @override
  void dispose() {
    _adapterStateStateSubscription.cancel();
    super.dispose();
  }

  @override
  Widget build(BuildContext context) {
    Widget screen;
    if (_adapterState == BluetoothAdapterState.on
        // || _adapterState == BluetoothAdapterState.turningOn
        // || _adapterState == BluetoothAdapterState.unknown
        ) {
      screen = const ScanPage();
    } else {
      screen = BluetoothOffScreen(
        bluetoothAdapterState: _adapterState,
      );
    }

    return MaterialApp(
      debugShowCheckedModeBanner: false,
      home: screen,
      navigatorObservers: [BluetoothAdapterStateObserver()],
    );
  }
}
```
Below is my flutter doctor output if it helps:

Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, 3.22.2, on Microsoft Windows [Version
    10.0.22621.3007], locale en-IN)
[√] Windows Version (Installed version of Windows is version 10 or higher)
[√] Android toolchain - develop for Android devices (Android SDK version
    34.0.0)
[√] Chrome - develop for the web
[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.8.4)
[√] Android Studio (version 2024.1)
[√] VS Code (version 1.91.0)
[√] Connected device (4 available)
[√] Network resources

• No issues found!

One interesting thing is that in Moto Edge 40, it's working, but in Vivo Z1 and Redmi 10 Prime, I'm getting the issue.