pauldemarco/flutter_blue #694: iOS issue: after connecting to a device once, it's never found again

Annelein, 2020-09-28:
Hi there,

I am building an app that connects with a bluetooth device (STM32 chip), and connecting and disconnecting to the device works for Android.

But with iOS this only works once and after disconnecting the device is never found again, this is also the case in a BLE scanner app. So it's not just my app that never finds it again.

Does anyone know how this is possible?
Or what might cause this issue (flutter/flutter_blue/the bluetooth device)?

I have absolutely no clue, also because it works for Android but just not for iOS...

thanks!

nikolabrdaroski, 2020-10-08:
@Annelein can you please try this:

`
 final FlutterBlue _flutterBlue = FlutterBlue.instance;

_flutterBlue.connectedDevices.asStream().listen((event) {
  // here should return already connected devices
}
`

Alex-Evo, 2020-11-04:
@nikolabrdaroski Is there a special reason why you suggest to use asStream().listen() instead of something like
devicesConnectedToPhone = await _flutterBlue.connectedDevices

nikolabrdaroski, 2020-11-04:
@Alex-Evo no, there is not special reason. Only thing is that I needed to listen for events when they become connected etc..