pauldemarco/flutter_blue #696: Plugin asks for bluetooth permission on app startup (iOS)

adamk22, 2020-10-08:
I'm using this plugin in an app where bluetooth is optional. However, in iOS, when the user boots up the app for the first time the plugin immediately asks for bluetooth permissions. Is it possible to control when to ask a permission? For instance, in a view where the bluetooth feature is utilized? 

I came across issue #272 with no clear solution other than digging into the source code and I was wondering if anyone has a solution?

MrCsabaToth, 2020-11-20:
Duplicate of #696

ollydixon, 2020-12-01:
Yeah this is a serious problem for us too. We'd like to be able to ask permission and explain why we need it too users before it actually asks them. Currentely it just pops up right away. This should be an easy fix, not sure how to implement it though (iOS)

anslex, 2020-12-06:
I agree that it is a serious issue to publish an app to market
Please improve this functionnality

MrCsabaToth, 2020-12-06:
Let's just first fix the even more serious issue of  #726

Chris-FizziQ, 2021-03-03:
Hi, I have this issue as well and it is quite unfortunate as it spoils the ergonomic of an app. Have any of you found some kind of solution ? Many thanks.

vaibhiarora03, 2021-04-24:
@Chris-FizziQ  check #599 .

Manasmd, 2023-07-31:
Do we have a fix from the package itself without having to change the IOS code for this?