pauldemarco/flutter_blue #700: flutter_blue example app not working on Flutter 1.20

AjayrajSingh, 2020-10-11:
Example app builds just fine with some API deprecated warnings, but get stuck on the blue screen, showing Bluetooth hardware not available, even Bluetooth is turned on. And if I try to scan for devices in another app using flutter_blue pub package, it never scans.

mprim, 2020-10-12:
Same problems here.

AjayrajSingh, 2020-10-13:
It's also showing flutter_blue API deprecated so it seems that the package hasn't upgraded to the latest version of flutter.

ljfass, 2020-10-29:
Same problems here.Version 1.22

goodxianping, 2020-11-06:
Same problems here.Version 1.22

tolk2go, 2020-11-12:
Hi Paul,

Do you plan to upgrade to the Flutter 1.22? If so, when?

Thanks, Jaap

bobosette, 2020-11-17:
> Example app builds just fine with some API deprecated warnings, but get stuck on the blue screen, showing Bluetooth hardware not available, even Bluetooth is turned on. And if I try to scan for devices in another app using flutter_blue pub package, it never scans.

Did you find any solution AjayrajSingh?

MrCsabaToth, 2020-11-20:
The API deprecation is one thing, I guess `flutter_blue` will be upgraded sooner or later but it should not cause the example to stop working. I think the reason for the bluetooth not available is #717