pauldemarco/flutter_blue #701: How to find my smart phone with flutter_blue?

martin284, 2020-10-12:
I'm very new to Bluetooth. I want to write a small app where I can test the functionalities of the package flutter_blue. I installed the official example here https://pub.dev/packages/flutter_blue/example on my tablet and tried to find my smart phone but it could not be found. It seems to be a stupid question but do I have to do something special so that my smart phone is recognized as a server by the tablet?

emretufekci, 2020-10-12:
This library does not support Classical Bluetooth instead of they're supporting BLE.

So, you need one central and one peripheral device

phones are central and with BLE technology it is not possible to connect to central device each other. So you need to transform one device to peripheral.

Use this library: https://pub.dev/packages/flutter_ble_peripheral

martin284, 2020-10-12:
Thank you for your helpful answer. I tried the example of the flutter_ble_peripheral package but still my phone can't be found. Are you sure that the package is necessary to just find (not connect) the phone?

emretufekci, 2020-10-12:
Yes it is necessary because of you need one peripheral point at least.

So one device should be central and other one should be peripheral. By using that package you can transform. If even you cannot see please make sure GPS & Bluetooth are enabled.

vipinkashyap, 2020-10-17:
You could instead install a bluetooth simulator app on a spare phone/device and use that for now as your beacon.

martin284, 2020-10-19:
Yes, I tried the BLE Peripheral Simulator and it works for me. Thanks.

jahnli, 2021-06-23:
> 是的，我尝试了 BLE 外设模拟器，它对我有用。谢谢。

I would like to know if you have implemented it and if you can contribute some code