pauldemarco/flutter_blue #707: PlatformException (PlatformException(connect, Peripheral not found, null, null))

joankabello, 2020-10-19:
Hello, 

I am using the plugin on an iOS 14 and this exception occurs when i kill the app and open it again.
Did anyone found a workaround?
I tried different workarounds in different issues and PR's that are not merged but it didn't fixed the problem.

Looks like a reinitialisation problem on Objective-C side when the app tries to find a Peripheral.

joankabello, 2020-10-20:
A workaround for this issue that i used on iOS devices.

If Platform.IsiOS, just make a scan with 2 seconds timeout and then use stopScan().

This happens because iOS doesn't save the connected devices list when the phone disconnects so the connect() will fail since the connected devices list is empty.

LEMUSADR000, 2021-03-03:
Hey, I saw that you created a fork of this library. When do you call the code to re-initialize?

joankabello, 2021-03-04:
Hey, its here
https://github.com/joankabello/flutter_blue/commit/70d77e7a295dde30a04765129983d2cff8aba3bb#diff-982e71202439d745e0d99cb29f213e29b587588b52fa929763de141d75b6df68R30

vaibhiarora03, 2022-01-22:
Hi @joankabello ,

Was this issue resolved fro you?

Thanks

JastySaran, 2023-11-11:
Please follow the below link the I have already answered to this question

https://github.com/pauldemarco/flutter_blue/issues/497#issuecomment-1805815352