pauldemarco/flutter_blue #709: It doesn't work properly on android10

wjy397, 2020-10-20:
It doesn't work properly on android10
Are you the same?

wjy397, 2020-10-20:
Adding this permission worked  : ACCESS_FINE_LOCATION

MrCsabaToth, 2020-11-20:
What's your app's Target API level? Android 10 and Android 11 have permission related changes.

MrCsabaToth, 2020-11-20:
ACCESS_FINE_LOCATION is part of https://github.com/pauldemarco/flutter_blue/blob/master/android/src/main/AndroidManifest.xml so your project inherits it, there's no need to add it as a duplicate, unless you perform any GPS location stuff or any Bluetooth operations outside of flutter_blue.

spekary, 2020-11-26:
I had to add ACCESS_FINE_LOCATION just to get debugging to work. Seems to be an Android 10 issue with flutter.

chiuhol, 2021-01-21:
Added ACCESS_FINE_LOCATION but didn't work

artemisja9, 2021-02-22:
Include this permission too for Android 10: ACCESS_BACKGROUND_LOCATION. If your app already granted the location permission, go into app info and allow location all the time, instead of just while app is in use.

MrCsabaToth, 2021-02-22:
> It doesn't work properly on android10
> Are you the same?

What do you mean by "It doesn't work properly"? What are the exact symptoms you are experiencing, what error messages do you get?