pauldemarco/flutter_blue #711: When I disconnect from device and call the FlutterBlue.instance.connectedDevices the disconnected device still appears.

anthonyorona, 2020-10-22:
// 1) Call device.disconnect();
           await device.disconnect();

    // 2) Call connected devices future in periodic stream (as shown in FlutterBlue examples)
    StreamBuilder(
        stream: Stream.periodic(Duration(seconds: 1)).asyncMap((_) => FlutterBlue.instance.connectedDevices),
        initialData: [],
        builder: (context, connectedSnapshot) {
            /// ...list open devices
        }
    )

If I do this, then the disconnected device still shows up in my stream. This leads me to believe that there is some asynchronous disconnection that we cannot synchronize on by calling await. Am I using the library wrong or is this the current expected behavior?

lifeinacastle, 2020-11-05:
I have the same issue, although I'm using a FutureBuilder.

narritt, 2020-11-26:
Also the device really still connected and I need to turn off Bluetooth to drop the connection.

AlexHartford, 2021-06-09:
Facing this same issue. Was anyone able to figure this out?

maddionatoplus, 2022-07-25:
did someone find a solution for this?