pauldemarco/flutter_blue #714: NullPointerException on Android

ekuleshov, 2020-10-29:
BLE scan sequence is failing with a NullPointerException on Android. It seems like `channel` is null (device got disconnected or something).

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void e.a.c.a.j.c(java.lang.String, java.lang.Object)' on a null object reference
       at com.pauldemarco.flutter_blue.FlutterBluePlugin$5.run(FlutterBluePlugin.java:10)
       at android.app.Activity.runOnUiThread(Activity.java:7095)
       at com.pauldemarco.flutter_blue.FlutterBluePlugin.invokeMethodUIThread(FlutterBluePlugin.java:7)
       at com.pauldemarco.flutter_blue.FlutterBluePlugin.access$400(FlutterBluePlugin.java)
       at com.pauldemarco.flutter_blue.FlutterBluePlugin$2.onScanResult(FlutterBluePlugin.java:83)
       at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:498)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:214)
       at android.app.ActivityThread.main(ActivityThread.java:7397)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)

```

MrCsabaToth, 2021-01-31:
I have this as well, only comes in released app, shows up in ANR time-to-time in Play Store.
What's confusing to me is `FlutterBluePlugin.java:10`. If you look at https://github.com/pauldemarco/flutter_blue/blob/master/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java those line numbers are off. However if we just follow the call stack from `onScanResult` maybe this is a Protocol Buffer V8/ProGuard issue? I've seen that with Firebase plugin.

adamburnet, 2021-02-01:
I have been seeing this in production as well. Looking at the stack trace, it looks like it's this call:

https://github.com/pauldemarco/flutter_blue/blob/c19bb9aadde250970353d22b3cd100db8bde28ff/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java#L998

This might be happening on startup when the activity isn't been registered for some reason. The activity object is not null checked here causing the exception (though a simple null check doesn't solve the issue).

https://github.com/pauldemarco/flutter_blue/blob/c19bb9aadde250970353d22b3cd100db8bde28ff/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java#L94

MrCsabaToth, 2021-02-01:
> I have been seeing this in production as well. Looking at the stack trace, it looks like it's this call: [...]

If that's the case then the root cause is deeper and more serious. With a simple null check at L149 you'd just prolong the inevitable crash, with a null Activity sooner or later you'd fail. The question is how the activity can be null. Maybe some race condition? Or the user changes orientation while starting? I don't have too many of these crashes, but they happen still every once in a while. I could never reproduce it on my end, and I believe this could be also one of those release-only bugs.

MrCsabaToth, 2021-03-10:
> I have been seeing this in production as well. Looking at the stack trace, it looks like it's this call: [...]

I was inspecting that call stack again, and can it happen that the `channel` is null at the `channel.invokeMethod(name, byteArray);`. The channel is a type of `MethodChannel` and is explicitly nulled out in the `tearDown` and is not initialized before `setup`. If we strictly look at the call stack the null pointer comes inside the `run` function **after** `android.app.Activity.runOnUiThread`

MrCsabaToth, 2021-03-10:
And I created a PR. I see this crash so regularly in the wild it is embarrasing. I thought about bringing back the `MainActivity.kt` (which I eliminated as part of https://github.com/flutter/flutter/wiki/Upgrading-pre-1.12-Android-projects) so I can override `runOnUiThread`. But `runOnUiThread` is `final`, so I cannot even do a pacman style catch there...

rtokun, 2021-05-05:
The same is here, we facing crashes in production when `channel` is null, while calling `channel.invokeMethod(name, byteArray);`.
From my check it happens, when an app goes to the background for a long time, then `onDetachedFromActivity` is being called and following `tearDown` cleaning the channel. Please merge the fix that was proposed by @MrCsabaToth .

MrCsabaToth, 2021-05-05:
> The same is here, we facing crashes in production when `channel` is null, while calling `channel.invokeMethod(name, byteArray);`.
> From my check it happens, when an app goes to the background for a [...]

That's great info. I've only seen this crash in the Play Store crashes, I was never able to reproduce it. The "when an app goes to the background for a long time" reproduction works for debug version or only for release?

iJamesPHP, 2021-06-18:
Seeing this same error cropping up in my Crashlytics, Android only. Any fix? Hoping it is just an extended background time bug and not something causing people big issues, thanks!

DivyanshuCaremonitor, 2021-06-28:
Got this issue only on prod. Is there any workaround?

[image]

MrCsabaToth, 2021-06-28:
> Got this issue only on prod. Is there any workaround?
> 

I'm seeing a crapton of these in production. I was never ever able to reproduce in debug, or even with release builds testing. I cannot wait to release a long waited update to my app which has my own flutter_blue fork.

DivyanshuCaremonitor, 2021-08-19:
Any update on this?

MrCsabaToth, 2021-08-19:
Here's my fork https://github.com/CsabaConsulting/flutter_blue

EliaTolin, 2022-05-08:
@MrCsabaToth Do you think your fork is still stable? Also how can I use it?

MrCsabaToth, 2022-05-08:
> @MrCsabaToth Do you think your fork is still stable? Also how can I use it?

This is how you refer a custom git repo within your pubspec: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/blob/cadd153f20277b8b8deb78e9088b8fa745873dd4/pubspec.yaml#L22

As you see I'm currently using my own fork still, but since flutter_blue got stuck in time, my fork is also not maintained and stuck in time. As Android 12 will take more market share it'll be more important to abandon flutter_blue. Within the upcoming months I gonna migrate over to `flutter_blue_plus` (by boskogs) and if that does not contain the Google ProtoBuffer related code exception catches, then unfortunately (and excrutiatingly painfully) I have to custom fork yet another f-ing time and rely on a custom fork again.

MrCsabaToth, 2022-05-08:
Here is my struggling: https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/210

EliaTolin, 2022-05-08:
> Here is my struggling: TrackMyIndoorWorkout/TrackMyIndoorWorkout#210

Would you recommend me to switch directly to flutter blue plus?  Have you already had experiences with it?  Is it stable?

MrCsabaToth, 2022-05-09:
> Would you recommend me to switch directly to flutter blue plus? Have you already had experiences with it? Is it stable?

I didn't have time to try it yet. But I also realize I don't have much more, other than another fork. I want to also tackle multiple issues, like the permission I mentioned, or Mac related ones by switching.

ekuleshov, 2022-05-09:
As far as I know `flutter_blue_plus` still haven't brought in macos support.

Also I don't think it is requesting permissions on android 12 and you'd have to use something else to request platform-specific permissions, perhaps `permission_handler` plugin.

I'm using this fork that at least addressed permission issues https://github.com/espresso3389/flutter_blue
Though macos support is one to one with the original - hence also broken.

@MrCsabaToth if you could propose and bring your changes in PRs to `flutter_blue_plus` that would be great. Especially if you fixed macos issues in yours.