pauldemarco/flutter_blue #717: Flutter_Blue 0.7.2 not compatible with Android X and Firebase Database - Protobuf Duplicate

larinnim, 2020-10-31:
Can't compile project running Android X and Firebase Database. I get "app:checkDebugDuplicateClasses" error.

> Task :app:checkDebugDuplicateClasses FAILED
         *********************************************************
WARNING: This version of flutter_blue will break your Android build if it or its dependencies aren't compatible with AndroidX.
         See https://goo.gl/CP92wY for more information on the problem and how to fix it.
         This warning prints for all Android build failures. The real root cause of the error may be unrelated.
         *********************************************************

FAILURE: Build failed with an exception.

Execution failed for task ':app:checkDebugDuplicateClasses'.
> 1 exception was raised by workers:
  java.lang.RuntimeException: Duplicate class com.google.protobuf.AbstractMessageLite found in modules protobuf-javalite-3.11.0.jar (com.google.protobuf:protobuf-javalite:3.11.0) and protobuf-lite-3.0.1.jar (com.google.protobuf:protobuf-lite:3.0.1)

kevin-induro, 2020-11-02:
In my case, this occurs with no library upgrades, just an upgrade from Flutter 1.20.4 to 1.22.x. I tried to exclude protobuf-lite, in my app/build.gradle file, but either it doesn't work, or I'm just using Gradle improperly (certainly likely).
```
implementation ('com.pauldemarco.flutter_blue:flutter_blue') {
	exclude group: 'com.google.protobuf', module: 'protobuf-javalite'	
	exclude group: 'com.google.protobuf', module: 'protobuf-lite'
}
```

DenizKucukozturk, 2020-11-04:
I have the same problem. In may case if the flutter project's android language is Kotlin everything is fine but with java I get the duplicate class errors. I need java version of android to implement some native android functionalities. Does anyone know any workaround ?

DenizKucukozturk, 2020-11-04:
There is this pull request which fixes protobuf related problems https://github.com/pauldemarco/flutter_blue/pull/676

narritt, 2020-11-08:
Just add 
`    
android {
    ...
    configurations {
        implementation.exclude module:'protobuf-lite'
    }
    ...
}
`
To your build.gradle in app module. Protobuf regular includes everything from protobuf-lite.

sdebasek, 2020-11-19:
I've added Flutter Blue to my project and was having this issue on dependencies below:
` cloud_firestore: ^0.14.3`
  `flutter_blue: ^0.7.3`

Turns out, when I have disabled flutter_blue the problems with "Execution failed for task ':app:checkDebugDuplicateClasses'.".

After some troubleshoting I found that changing `\android\app\build.gradle` `minSdkVersion 19` (which is recommended by flutter_blue) to `minSdkVersion 21` fixed the issue for me.

Maybe it will help someone.

kevin-induro, 2020-11-20:
@sdebasek I'm (and was) on a similar version of Firestore, and my minSdkVersion was already 21. I'm was on 0.7.2 of flutter_blue, but I have moved over to flutter_ble_lib since this was a show stopper.

sdebasek, 2020-11-20:
The problem was fixed on 0.7.3 (and was present on 0.7.2 no matter what minSdkVersion you would choose), but still occurs when using Firebase/Firestore plugins - then only minSdkVersion higher or equal to 21 solves the issue for now.

kevin-induro, 2020-11-20:
@sdebasek I see that now. I did not have confidence in the PR getting tested and approved so quickly! Good to know.

Should this issue then be closed?

MrCsabaToth, 2020-11-24:
> @sdebasek I see that now. I did not have confidence in the PR getting tested and approved so quickly! Good to know.
> 
> Should this issue then be closed?

For me the #700 is present (a.k.a. "get stuck on the blue screen, showing Bluetooth hardware not available, even Bluetooth is turned on.") with 0.7.3 and when I revert back I don't have it. Only in release versions. So there's still a bug somewhere.

aawhitfield, 2020-11-26:
I updated to 0.7.3 because of the incompatibilities with Firebase. Now when I try to build a release version, I get: 
`FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* Where:
Build file 'C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.7.3\android\build.gradle' line: 37

* What went wrong:
A problem occurred evaluating project ':flutter_blue'.
> Failed to apply plugin [id 'com.google.protobuf']
   > Gradle version is 5.4.1. Minimum supported version is 5.6

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':flutter_blue'.
> compileSdkVersion is not specified.

`