pauldemarco/flutter_blue #719: I am getting different value for `device.id` in ios and android. Solution?

zeeem, 2020-11-06:
**I am trying to connect a BLE device depending on its `device id` or `MAC address`.**

Using same flutter project-
- When I search using an **android device** I get the device id: `66:55:44:33:22:11`
- And on the **IOS device**, for the same BLE device, I am getting the device id: `F1547A55-0823-261A-B1C2-051234E9DAE9`

The device name is the same, but the device id changes on different platforms. 
How can I solve it?

muhammadfaheem-pycom, 2020-11-10:
any solution for this?

MBjoern, 2020-11-11:
You can't! This is not a Flutter_Blue issue, please read the dedicated documentations of iOS.
iOS is obfuscating the BLE mac address of gadgets, hence the cryptic device Id. It's randomly generated and also differs from one iOS device to another. You won't see the same device Id on different iOS devices.
The Id you see on Android is the actual BLE mac address.
There is no way, to get the mac address on an iOS device. If you need a device identifier, you have to add it to the advertising data of the gadget or add a dedicated characteristic exposing the mac address yourself.

zeeem, 2020-11-16:
Ohh I see.  :(
Thanks.

MrCsabaToth, 2020-11-20:
I assume this can be closed then.

closed 2020-11-20 by zeeem

EdwardRodr1guez, 2023-11-01:
> You can't! This is not a Flutter_Blue issue, please read the dedicated documentations of iOS. iOS is obfuscating the BLE mac address of gadgets, hence the cryptic device Id. It's randomly generated [...]

@MBjoern even if the mac is exposed via a dedicated characteristic, isn't a little chance that the OS detect it and then, ofuscate it too?