pauldemarco/flutter_blue #725: How to scan for ever

ControlAdad, 2020-11-17:
Hello
How can I use flutter blue to detect when I enter my car and my phone connects to my car (they have paired already).
I want to use it to automatically start music playback in my app.
Is it logical to scan for ever:
flutterBlue.startScan(timeout: Duration(seconds: 9999999999));
Thanks

Chakib-Temal, 2020-11-24:
Hi , did you have any solution ? 
they should add a function like onScanFinished() to restart a new Scan

Chakib-Temal, 2020-11-24:
after some search, Here is my solution
in Dart Language , we can use a Stream object 
 
```
      @override
       void initState() {
               setStream(getScanStream());
      }

      Stream<ScanResult> getScanStream() {
          return FlutterBlue.instance.scan(timeout: Duration(seconds: 10));
      }

      void setStream(Stream<ScanResult> stream) async {
        stream.listen((event) {
            print("data received $event");
        }, onDone: () async{

            // Scan is finished ****************
            await FlutterBlue.instance.stopScan();
            print("Task Done");
            print("second scan");
            setStream(getScanStream()); // New scan

        }, onError: (Object e) {
            print("Some Error "  + e.toString());
        });
  }

```

febryardiansyah, 2020-11-29:
Hi, if you want to scan repeatedly. u must not use duration, just use `flutterBlue.startScan();`

ram4nd, 2021-02-15:
@Chakib-Temal Thank you for this solution. This is really good, you have a lot more control. Also `flutterBlue.startScan();` will stop working when you manually turn bluetooth on and off.

gordett, 2021-12-25:
I try flutterBlue.startScan without a duration, and it passes a lot of time and it doesn't scan new devices... 
any reason?