pauldemarco/flutter_blue #726: Bluetooth adapter is not available

bobosette, 2020-11-17:
Hi everybody. I'm trying the flutter blue example on an Android 10 device but when I open the app i get this msd: "the bluetooth adapter is not available"! 
If I debug that I see that the FlutterBlue.instance.state is null! But if i open the app in an older Android device it works fine!
Can somebody help me?

Hoopstr42, 2020-11-18:
I am getting the same issue in my project but only when do a release build. 

I have run it on Android 10 & 11 same issue. 

Android Manifiest
`<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.flutter_base">
    
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.RECORD_AUDIO" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
</manifest>`

build.gradle
`android {
    compileSdkVersion 29

    lintOptions {
        disable 'InvalidPackage'
    }

    defaultConfig {
        applicationId "com.x.x"
        minSdkVersion 24
        targetSdkVersion 29
        versionCode flutterVersionCode.toInteger()
        versionName flutterVersionName
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
    }`

bobosette, 2020-11-18:
I didn't think about it but I also have this issue only in a release build. How is it possible??

JimmyLee05, 2020-11-18:
I have the same issue

ch-muhammad-adil, 2020-11-18:
Please try to role back to 0.7.2 and also make sure you have removed these permissions because these are already there in the plugin. 

```
android.permission.ACCESS_FINE_LOCATION
android.permission.BLUETOOTH
android.permission.BLUETOOTH_ADMIN
```

Hoopstr42, 2020-11-18:
Just FYI, we found a workaround by rolling back to v0.7.2. So if you don't need the protobuf-lite fix. Just roll back and you will be good to go.

bobosette, 2020-11-18:
Ok i'm gonna try this and i'll let u know

bobosette, 2020-11-18:
You're right. Now the app works correctly. Thank you.

JimmyLee05, 2020-11-19:
Please try " flutter build apk --no-shrink " to build your release build.

bufolab, 2020-11-19:
Yes, 0.7.3 with "flutter build apk --release" is not working.  With "flutter build apk --no-shrink " indeed works.
Looks like R8 is messing up.

MrCsabaToth, 2020-11-20:
> Just FYI, we found a workaround by rolling back to v0.7.2. So if you don't need the protobuf-lite fix. Just roll back and you will be good to go.

I just hit this, only release build is affected. Is there an other issue involved? Can your reference? I don't need or use protobuf.

MrCsabaToth, 2020-11-20:
Ok, I think this is a protobuf issue @Hoopstr42 mentions: https://github.com/pauldemarco/flutter_blue/issues/717
I saw the "WARNING: This version of flutter_blue will break your Android build if it or its dependencies aren't compatible with AndroidX.".

ayushin, 2020-12-02:
We should mention this as a breaking change.

MrCsabaToth, 2020-12-16:
Guys, I nuked my whole android port and re-created it because of https://github.com/miguelpruivo/flutter_file_picker/issues/545 and I tested and now 0.7.3 does not produce the release-only error for me. This means that this problem is not with flutter_blue but because of the complexity and interference of this system with many moving parts.

MrCsabaToth, 2020-12-27:
Lately even nuking and recreating the android port doesn't solve this.

MrCsabaToth, 2020-12-28:
https://github.com/pauldemarco/flutter_blue/issues/662 helped

JAICHANGPARK, 2021-01-05:
I have same issue

chanum, 2021-01-07:
flutter_blue: ^0.7.3

I've the same issue building the APK release with **"flutter build apk --release"** in Android 7.1.1 (Samsung Tablet) and Phone (Motorola Android 10). 

But the workaround **"flutter build apk --no-shrink"** works for me.

JAICHANGPARK, 2021-01-08:
> flutter build apk --no-shrink

this is not works for me (android 8.1)

GuruJohn, 2021-01-15:
Seems to be happening on IOS as well, and none of the work arounds function.

h582618, 2021-03-12:
Still an issue in IOS, none fix found so far.

anthonyorona, 2021-07-06:
@MrCsabaToth Hey small world I saw you at Google Dev Fest in Fresno.. maybe like 2017?

MrCsabaToth, 2021-07-07:
> @MrCsabaToth Hey small world I saw you at Google Dev Fest in Fresno.. maybe like 2017?

Oh yes, Valley DevFest!

myogesh7, 2022-01-17:
Work-around for https://github.com/pauldemarco/flutter_blue/issues/772
add the below two lines in buid.gradle(app) -> **buildTypes** -> release 
            shrinkResources false
            minifyEnabled false

[image]
 
 works like magic

Githubsaveme, 2022-07-21:
buildTypes {
        release {
            // TODO: Add your own signing config for the release build.
            // Signing with the debug keys for now, so `flutter run --release` works.
            signingConfig signingConfigs.debug
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
            shrinkResources false
            minifyEnabled false
        }
    }

Note:- Run  **flutter build apk --no-shrink**

Working :)