pauldemarco/flutter_blue #732: Not working on release mode, but works on debug mode.

yun-li, 2020-11-27:
I found some interesting things with flutter blue. 

Flutter blue works when I connect a device to my computer and run
```flutter run --release```
or
```flutter run --debug```

It also works when I build the apk with
```flutter build apk --debug``` and then install with apk on my device.

but if I build the apk with
```flutter build apk --release``` and install the generated release apk on my phone, flutter blue can not detect any devices. 

The scan call back 
```var subscription = flutterBlue.scanResults.listen((results){}```
would only be called once with an empty `results` list.

Do anyone experience such an issue?

Thanks!

realrk69, 2020-11-27:
I'm experiencing the opposite, it works on 'run' mode of Android Studio, but not on 'debug'. Honestly, quite a lot of a pain to debug.

yun-li, 2020-11-27:
Found not to be a problem from flutter blue.
It works if I do `flutter build apk --no-shrink`

Flutter is doing something silly when shrinking the code. (I have a java plugin)

closed 2020-11-27 by yun-li