pauldemarco/flutter_blue #734: When will NNBD arrive?

aabounegm, 2020-11-28:
Are there any plans to migrate the package to null-safety soon? All the dependencies now seem to offer null-safe versions

shinayser, 2021-02-12:
@pauldemarco can you take a look on this, please?

savs90, 2021-03-04:
Any updates on this one guys ?

jlubeck, 2021-03-04:
Null safety is now on stable branch since yesterday, so it would be nice to have this so we can convert out projects to fully null safety. Thanks!

safield, 2021-03-04:
Guys I don't know why no one has put 2+2 together yet, but this library is not actively maintained.

Migrate to one of the following...

https://github.com/PhilipsHue/flutter_reactive_ble
https://github.com/Polidea/FlutterBleLib

jlubeck, 2021-03-04:
Thank you @safield , made the comment hoping it would get some traction, and you comment gave me traction

spiderion, 2021-03-14:
flutter_blue:
    git:
      url: https://github.com/spiderion/flutter_blue_null_safety

navaronbracke, 2021-03-16:
Version 0.8.0 now has Null Safety.

closed 2021-03-17 by aabounegm