pauldemarco/flutter_blue #736: Device takes a long time to connect (sometimes)

realrk69, 2020-11-29:
I'm trying to connect an nRF device using Flutter Blue. I'm using a very simple implementation as of now, here is the error log and a runnable code:
Error Log:
```
D/BluetoothAdapter(32340): isLeEnabled(): ON
D/BluetoothGatt(32340): connect() - device: B8:27:EB:98:60:9B, auto: true
D/BluetoothGatt(32340): registerApp()
D/BluetoothGatt(32340): registerApp() - UUID=29853642-46a3-48f5-b5e9-72eb2d84c68d
D/BluetoothGatt(32340): onClientRegistered() - status=0 clientIf=5
D/BluetoothAdapter(32340): isLeEnabled(): ON
D/BluetoothLeScanner(32340): could not find callback wrapper
```

realrk69, 2020-12-04:
Shifted to flutter_reactive_ble. The owners frequently patch things up. This library really needs some optimization. Would re-explore then.

closed 2020-12-04 by realrk69

JAICHANGPARK, 2021-01-21:
@pauldemarco I had same problem about this 
any have this solution?