pauldemarco/flutter_blue #737: is there is a method to pair a device to another device running same app?

mohadel92, 2020-11-30:
hi,
is there is a method to pair a device to another device running same app with flutter_blue package installed
so that they can send and receive data without connecting every time to each other
like normal android phone they automatically connect to each other when bluetooth is on and they have already paired succesfully

realrk69, 2020-11-30:
This library is for normal BLE connection. What you're looking for is HC-05. It can be done on this library also, but I would not recommend as you'll have to store the UUIDs, characteristics and Mac address which can be changed (or create your own), and that will cause lots of errors.

mohadel92, 2020-12-01:
@realrk69 can i have a further explanation ,or where to dig after this topic

edilson14, 2020-12-01:
i have the same problem , any luck @mohadel92 ?? 
first can you provide some doc/link about the pairing ?

mohadel92, 2020-12-01:
@edilson14 no luck till now
the documentation regarding flutter_blue is very poor towards this point
i think doing such an activity require profound understanding for what's going under the hood .
right now i am digging after this topic natively @edilson14

edilson14, 2020-12-01:
@mohadel92  i am going to test this [library](https://pub.dev/packages/flutter_bluetooth_serial) only for pairing feature 
maybe this can help us https://pub.dev/documentation/flutter_bluetooth_serial/latest/flutter_bluetooth_serial/BluetoothPairingRequest-class.html
https://pub.dev/documentation/flutter_bluetooth_serial/latest/flutter_bluetooth_serial/PairingVariant-class.html
https://pub.dev/documentation/flutter_bluetooth_serial/latest/flutter_bluetooth_serial/BluetoothPairingRequest/BluetoothPairingRequest.html

edilson14, 2020-12-01:
> @mohadel92 i am going to test this [library](https://pub.dev/packages/flutter_bluetooth_serial) only for pairing feature
> maybe [this](https://github.com/edufolly/flutter_bluetooth_serial/issues/67#issuecomment-525694508) can help us
> 
> `import 'package:flutter_bluetooth_serial/flutter_bluetooth_serial.dart' as pairing; . . . . pairing.FlutterBluetoothSerial.instance .bondDeviceAtAddress(device.id.toString(), pin: '012345')`

@mohadel92 this work for me :)

mohadel92, 2020-12-02:
@edilson14 will the apps reconnect to each other if they came closer again ?
or do i have to reconnect them?

edilson14, 2020-12-02:
> @edilson14 will the apps reconnect to each other if they came closer again ?
> or do i have to reconnect them?

i do not have the answer for that question , but the reconection is a feature for this library