pauldemarco/flutter_blue #738: Trying to receive data in Android

realrk69, 2020-11-30:
I run this method:

```
getBackgroundFunction(final List<int> writeBgCommand) async {
    await _tx.write(writeBgCommand, withoutResponse: false);
    List<int> value = await _rx.read();
    print(value);
  }
```
The value shows as null. It's supposed to be 20 packets of List<int> which are received as a response to writing the characteristic. The device responds correctly.

```
getBackgroundFunction(final List<int> writeBgCommand) async {
    List<int> value = await _tx.write(writeBgCommand, withoutResponse: false);
    print(value);
  }
```
This also returns as null. What am I doing wrong here? The characteristics are setup correctly...

realrk69, 2020-11-30:
Solved it. I had to read notify events. Thanks a ton to the developers. Let me know if you guys need some help in implementing features. I'd love to see this library evolve.

closed 2020-11-30 by realrk69

vaibhiarora03, 2020-12-19:
I am also facing issues with receiving the data. In native iOS I used to write and read via a single characteristic but when I tried doing the same in flutter, I got no response. Then after reading your above comment, I set reading to all the characteristics received from all the services. 

Also, when received the required data from another characteristic, it's unable to get the decoded value because when I'm trying to decode the value it gives me null and and the value itself is not printed in the console. To me it seems like when decoding the value received from the characteristic creates issues to the Read functionality. And upon commenting the decoding part, the value is received and printed from the characteristic.

This is how I start discovering the devices.

`@override
  void initState() {
    // TODO: implement initState
    widget.flutterBlue.scanResults.listen((List<ScanResult> results) {
      for (ScanResult result in results) {
        _addDeviceTolist(result.device);
      }
    });
    widget.flutterBlue.startScan();
  }`

vaibhiarora03, 2020-12-19:
I am also facing issues with receiving the data. In native iOS I used to write and read via a single characteristic but when I tried doing the same in flutter, I got no response. Then after reading your above comment, I set reading to all the characteristics received from all the services. 

Also, when received the required data from another characteristic, it's unable to get the decoded value because when I'm trying to decode the value it gives me null and and the value itself is not printed in the console. To me it seems like when decoding the value received from the characteristic creates issues to the Read functionality. And upon commenting the decoding part, the value is received and printed from the characteristic.

This is how I start discovering the devices.

`@override
  void initState() {
    // TODO: implement initState
    widget.flutterBlue.scanResults.listen((List<ScanResult> results) {
      for (ScanResult result in results) {
        _addDeviceTolist(result.device);
      }
    });
    widget.flutterBlue.startScan();
  }
` 

Connect Device:-

`
void _connectDevice() async {
    widget.flutterBlue.stopScan();
    print(_deviceToConnect);
    try {
      await _deviceToConnect.connect();
    } catch (e) {
      if (e.code != 'already_connected') {
        throw e;
      }
    } finally {
      _services = await _deviceToConnect.discoverServices();
    }
    _discoverDeviceServices();
    setState(() {
      _connectedDevice = _deviceToConnect;
    });
  }
`

Discovering the services:-

Commenting TAG-1 & TAG-2 lines gave me the values which I need from the characteristic. I tried both ASCII and UTF8 decoding. 
`
void _discoverDeviceServices() async {
    for (BluetoothService service in _services){
      for (BluetoothCharacteristic characteristic in service.characteristics){
        var value = await characteristic.read();
        print(value);
        // print(AsciiDecoder().convert(value)); /*** TAG-1***/
        // print(utf8.decode(value)); /*** TAG-2***/
        if (characteristic.properties.write){
          if (characteristic.properties.notify){
            _rx_Write_Characteristic = characteristic;
            _sendCommandToDevice();
          }
        }
      }
    }
  }
`

Writing commands to device:-

`void _sendCommandToDevice() async {
    final command = "AT Z\r";
    final convertedCommand = AsciiEncoder().convert(command);
    await _rx_Write_Characteristic.write(convertedCommand);
    _connectedDevice.disconnect();
  }`

Can you please help me out pointing where I got wrong in reading the values from the characteristics?

Thanks!

vaibhiarora03, 2020-12-19:
`void _sendCommandToDevice() async {`
`    final command = "AT Z\r";`
`    final convertedCommand = AsciiEncoder().convert(command);`
`    await _rx_Write_Characteristic.write(convertedCommand);`
`    _connectedDevice.disconnect();`
  `}`

vaibhiarora03, 2020-12-19:
> getBackgroundFunction(final List<int> writeBgCommand) async {
    await _tx.write(writeBgCommand, withoutResponse: false);
    List<int> value = await _rx.read();
    print(value);
  }

eliseiml, 2021-06-09:
> Solved it. I had to read notify events. Thanks a ton to the developers. Let me know if you guys need some help in implementing features. I'd love to see this library evolve.

Hi! Couldn't you explain more explicitly or show your code example?

realrk95, 2021-06-09:
> > Solved it. I had to read notify events. Thanks a ton to the developers. Let me know if you guys need some help in implementing features. I'd love to see this library evolve.
> 
> Hi! Couldn't you explain more explicitly or show your code example?

We actually have shifted to flutter_reactive_ble. Much more stable and compatible with all devices on Android and iOS

eliseiml, 2021-06-09:
> We actually have shifted to flutter_reactive_ble. Much more stable and compatible with all devices on Android and iOS

Thanks, I gonna try it too

Logamuhilan, 2021-12-11:
Hi @realrk95 i need your help now.
I am using flutter_blue to connect and get Bluetooth device data, my problem is i don't know the method to get data from it. I have used "await c.read" it gives only device name, and "await d.read" it gives only empty array. here "c" and "d" is characteristics. I think read is not a method i think. Is there any method to get data from my device? Cheers in advance ; )