pauldemarco/flutter_blue #742: Where we can find UUID, Major and Minor values

bambinoua, 2020-12-04:
We are using you package but we cannot find the required information. Where we can find UUID, Major and Minor values? The `advertisementData` it looks like does not contains this info.

bambinoua, 2020-12-04:
Well, it looks like `device.id.id` is the beacon UUID. Correct? But where to find Major and Minor values?

bambinoua, 2021-01-08:
Author, could you please to answer?

qurm, 2021-01-08:
I dont think that is built in to the library.  You would need to get that that from parsing the UUID.  There some lines in the example app like this (for service, but similar for device).

`            Text('0x${service.uuid.toString().toUpperCase().substring(4, 8)}',`

bambinoua, 2021-01-17:
It seems I don't understand. UUID is UUID, Major number is major number. Minor number is minor number. All these are three separated characteristics of beacon so what relation between UUID and major/minor you talking about? Where do you take the formula for major/minor extraction?
P.S. We started to use another plugin which contains required data out of the box.

ibcoleman, 2021-07-26:
> P.S. We started to use another plugin which contains required data out of the box.

Hey @bambinoua! Which library did you end up using? I agree getting simple identity info from the library shouldn't be some kind of moon-shot.