pauldemarco/flutter_blue #743: Release apk doesn't work

Aleja14, 2020-12-07:
When i release my apk, flutter_blue doesn't work. In debug mode work fine.

naxiemolv, 2020-12-08:
+ 1,same issus

sami-kouatli-act, 2020-12-09:
Same issue here. Works fine in debug but not in release. Anyone has a fix ?

wcoder, 2020-12-10:
Seems like a workaround: https://github.com/pauldemarco/flutter_blue/issues/662#issuecomment-741710564

sami-kouatli-act, 2020-12-11:
> 
> 
> Seems like a workaround: [#662 (comment)](https://github.com/pauldemarco/flutter_blue/issues/662#issuecomment-741710564)

Thank you, it works now in release.

narritt, 2020-12-21:
It's working, but this is a very bad workaround because release versions need shrinking and minify.

JAICHANGPARK, 2021-01-05:
I had a same problem

Artur110, 2021-01-21:
Same here, pleas fix it

groothuisss, 2021-01-21:
Same issue for me on version 0.7.3 with various Android versions. Works fine in debug mode, but not in release mode.

devfelipereis, 2021-01-28:
Same here. Workaround works.

MaykonDS, 2021-02-10:
Here, i switch the flutter channel to stable and resolved, in beta i was with the same problem, debug works but not in release.

RafaRuiz, 2021-04-13:
For those who need "shrinking" and obfuscation, if you add the following line to `proguard-rules.pro`

```
-keep class com.pauldemarco.flutter_blue.** { *; }
```

it'll work and the rest of your app will be correctly obfuscated as well.

groothuisss, 2021-04-13:
The suggestion of @RafaRuiz worked for me, but I had to add #noinspection ShrinkerUnresolvedReference to prevent an inspection error about not being able to find the package.

For completeness, my build.gradle: 

`buildTypes {
        debug {
            signingConfig signingConfigs.debug
        }
        release {
            signingConfig signingConfigs.release
            shrinkResources true
            minifyEnabled true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }`

My proguard-rules.pro in <projectname>/android/app/proguard-rules.pro:

`#noinspection ShrinkerUnresolvedReference
-keep class com.pauldemarco.flutter_blue.** { *; }`

Hope it helps, cheers!