pauldemarco/flutter_blue #750: scan() returns devices ids but no names

darre1, 2020-12-20:
Hi

I am using the following code to populate a `_devices` variable (a list of ScanResults) which is then displayed on-screen. It works fine but I just get a list of devices with ids, rssi etc and the name is always blank for each device.

Any ideas how to get the name of a device (i.e. what it is 'discoverable' as)?

```dart
// list devices on-screen as they are found (i.e. in real-time) during the scanning period
await for (ScanResult r in FlutterBlue.instance.scan(timeout: Duration(seconds: 5))) {
  setState(() async {
    _devices.add(r);
  });
}
```

yukitaka13-1110, 2021-03-19:
Any updates about this?

xhp281, 2021-03-28:
> Hi
> 
> I am using the following code to populate a `_devices` variable (a list of ScanResults) which is then displayed on-screen. It works fine but I just get a list of devices with ids, rssi etc and the name is always blank for each device.
> 
> Any ideas how to get the name of a device (i.e. what it is 'discoverable' as)?
> 
> ```dart
> // list devices on-screen as they are found (i.e. in real-time) during the scanning period
> await for (ScanResult r in FlutterBlue.instance.scan(timeout: Duration(seconds: 5))) {
>   setState(() {
>     _devices.add(r);
>   });
> }
> ```

I also encountered this problem, how to solve it?

gumbypp, 2021-03-29:
Initial scans, at least on iOS, don't always include the name so you might need to wait for a `scanResult` that does. In my case, I am scanning for peripherals that advertise a specific service and their device names also start with a certain prefix. If the names of devices that you're scanning for don't start with a common prefix, you might just be able to check for a non-zero length if `scanResult.advertisementData.localName` is not `null`.

```
var scanSubscription = _flutterBlue.scan(withServices: [ kSpecificService ]).listen((scanResult) {
    final String deviceName = scanResult.advertisementData.localName;
    debugPrint("discovered '$deviceName' with rssi ${scanResult.rssi}");

    // initial BLE scans sometimes don't get the advertised name
    if (deviceName.startsWith(kDeviceAdvNamePrefix)) {
        callback(deviceName, scanResult.rssi);
    }
});
```