pauldemarco/flutter_blue #754: Hot restart results in scan that yields no results.

safield, 2020-12-22:
I am calling the following code...

```
      _flutterBlue.scanResults.listen(_scanResultsListener); 
      _flutterBlue.startScan(timeout: SCAN_DURATION);
```
When first launching the application with flutter run, the _scanResultListener method get passed a ScanResults list that contains the expected devices.

When I then run a "Hot restart" to restart the application, _scanResultsListener is not passed a List containing the expected ScanResults.

If I then kill flutter run, and start from the beginning, FlutterBlue yields the expected ScanResults.

tk512, 2021-01-02:
This confused me as well, while developing a Flutter app on Android using FlutterBlue. There seems to be some stuck state in either FlutterBlue's internal state or some kind of a bug in the Bluetooth subsystem on Android (less likely).

For now the workaround is to stop the application (flutter kill or hit 'stop' in Android studio) every time you play around with FlutterBlue listeners and such.

vaimikpatel2908, 2021-03-02:
I am not sure what was the problem but I faced a bit similar kind of problem. Plugin Stopped scanning for the Bluetooth devices. I have tried several solutions like flutter clean, rebuilding an application, and restarting a code editor(vs code and android studio) as well. But nothing worked out. After 2-3 hours of effort, I gave up and then thought to restart my testing device,  and to my surprise, it worked out. Therefore, not sure whether it is the problem on the plugin side or the Android API side. 

I was not able to reproduce a problem after that, but for information, if someone wants to dig into this problem:

Device: Google Pixel XL
OS: Android 10
Flutter Blue Version: 0.7.3(Latest)