pauldemarco/flutter_blue #757: Sending Quite Large File from BLE device to App

bobosette, 2020-12-31:
Hi everybody. I need to send a file .txt from a device to my app (worst case almost 2mb). The BLE device divide the file into packages. I don't know if my method is correct, but I create a loop of characteristic.write/characteristic.read telling everytime what package the device has to send.
Here's my code:

`
for(int i = 0; i < packNumber.length; i++) {
  

  initialValue = '9,50,100,$i,0,$checksumId,0/';
                                                            
  await characteristic
  .write(utf8.encode(
      initialValue))
  .then(
      (wValue) async {
            await Future.delayed(Duration(milliseconds: 100)).then((value) async {
              await characteristic.read().then((rValue) {
                //do something with rVlaue
              });
            });
  });
}
`
It works, but is it the best solution? And in case how can I speed up the transfer (for now I have to set a delay before reading, waiting for characteristic.write to finish)?

Thank you guys

bobosette, 2021-01-02:
Nobody has any suggest? I need to know if I am on the right way or not

zlj-zz, 2021-01-07:
I tried to do this, but there was a packet loss anyway. 
It's hard to get all the packets. 

Note: my file size is about 20m

bobosette, 2021-01-07:
I don't have a packet loss with withoutResponse:false, but is not so fast

KevinRohn, 2021-01-08:
Maybe this repo helps you:
https://github.com/KevinRohn/flutter_ble_data_handler

bobosette, 2021-01-12:
Thank I will try it.

To optimize the transfer instead of using c.read i tried with c.value.listen(....) which detect any values changing. Maybe is the right way, but why packages I receive with c.read are 64byte long and packages with c.value.listen are maximum 20byte long??