pauldemarco/flutter_blue #760: Can't discover services of a "paired" device on iOS

Hoopstr42, 2021-01-14:
The issue is that once a BLE device is "paired" via iOS in the system you can not get services for that device once you close and reopen the app.

We have done a ton of troubleshooting and investigating and think we found the root of the issue. 

Steps to recreate:
Open app first run > connect to ble device > discover services everything works fine

iOS asks you to pair the device > device is paired 

Close App

Open App second time > BLE Device is already connected by iOS before the app opens
- when you call  _bluetooth.connectedDevices.asStream().listen((devices) the device is returned as a connectedDevice
-  if you then call device.state.listen((state) > it returns BluetoothDeviceState.disconnected
- if you call device.discoverServices() > you get an error that says - "Cannot discoverServices while device is not connected. State == BluetoothDeviceState.disconnected" - this due to a check of state on line 58, in bluetooth_device.dart

We know that the device is actually connected because: 
1. you can see it is connected in the iOS ble menu.
2. It is not present when a scan of ble devices is done in the app or with an outside scanner app. 
3. We tried to device.connect() but it always times out

After investigation of the Objective-C code I think we found the issue:

BluetoothDeviceState only updates during the transition of state > connecting, connected, disconnected, etc. But this never happens because the device is already in the connected state when the app opens and the default state for BluetoothDeviceState is disconnected. 

Our fix we suggest and are working on is to also do a check (on native side) for each device returned in the connectedDevices() method as these devices come straight from the OS and we can guarantee that they are connected. The do a check to make sure each device returned is set to BluetoothDeviceState.connected.

It would be great to get some direction on this or even a hot fix as this is causing a huge paint point with our customers.

Thanks in advance.

here is the code we are running

`_bluetooth.state.listen((state) async {
      bleState = state;
      print('LISTEN - STATE $bleState');
      _bluetooth.connectedDevices.asStream().listen((devices) async {
        print('LISTEN - DEVICES $_connectedDevices');

        for(BluetoothDevice device in _connectedDevices){
          device.state.listen((state) {
            print('1 - DEVICE STATE $state');  // always .disconnect
            
            if(state == BluetoothDeviceState.connected) {
              //never runs
              device.discoverServices();
              connectToCharacteristics(services);
            }
            if(state == BluetoothDeviceState.disconnected) {
              connectToDevice(savedBLE); // always timesout
            }
           device.discoverServices();  // "Cannot discoverServices while device is not connected. State == BluetoothDeviceState.disconnected"
          });
        }
      });
    });`

Hoopstr42, 2021-01-16:
I see this in flutter_blue.dart -> 
[image]

It looks like you know about this issue. Hoping you can let me know what to do.

if you do device.connect() it is never returned because the device is already connected.

JonYun, 2021-09-25:
I also encountered the same needs as you. I would like to know if you have solved this problem?

jojoX94, 2023-04-14:
Same issue

jojoX94, 2023-04-14:
Same issue

RafCoder90, 2024-06-19:
In my app, I scan, connect to the device, and discover services.
Android everything is ok. 
But in IOS:

- scan device OK
- connect to device OK
- discovery services ZERO SERVICES FOUND