pauldemarco/flutter_blue #764: device connect : status: 133 newState: 0

JAICHANGPARK, 2021-01-21:
I have some error about device connect process 
> [onConnectionStateChange] status: 133 newState: 0

any have solved this problem?

JAICHANGPARK, 2021-01-21:
after multiple connection 

```
D/BluetoothManager(28862): getConnectionState()
D/BluetoothManager(28862): getConnectedDevices
D/BluetoothManager(28862): getConnectionState()
D/BluetoothManager(28862): getConnectedDevices
```

ollex, 2021-02-19:
Hi @JAICHANGPARK - the 133 error is very common on Android devices, and it seems like this is related to the underlying bluetooth stack on each device and you can't really fix it from your own code; so usually the best way to deal with it is to just accept this error a few times, and try to reconnect anyway without letting this bubble up all the way until the user will recognize it. If it fails say 3 times in a row to connect in your connection loop trying to get a successfull connect out of 3-4 attempts, you can inform the user that he should try to delete the bluetooth cache by for example turning bluetooth off for a short period and on again. Most of the times this will help. On very old Androids they might even have to turn the phone off and on again, which will help it to work for some time. This was the only measure to kind of resolve the issue for me. On IOS this seems to be handled (if it happens at all) by the bluetooth stack, so usually on IOS only one connection attempt is needed and it is successfull most of the times.

JAICHANGPARK, 2021-02-19:
Hi @ollex 👋 awesome! 
Thank you for comment