pauldemarco/flutter_blue #765: Duplicate classes when using flutter_blue and firebase

EncryptedSpirit, 2021-01-23:
# Issue
I'm getting a lot of classes duplicated. I think that it because I'm using firebase.

## Example
Duplicate class com.google.protobuf.SmallSortedMap found in modules protobuf-javalite-3.11.0.jar (com.google.protobuf:protobuf-javalite:3.11.0) and protobuf-lite-3.0.1.jar (com.google.protobuf:protobuf-lite:3.0.1)

## Version
flutter_blue: ^0.7.3

DeveloperJ11, 2021-03-19:
hello friend, have you been able to solve this problem? I have the same problem

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugDuplicateClasses'.
> 1 exception was raised by workers:
  java.lang.RuntimeException: Duplicate class com.google.protobuf.AbstractMessageLite found in modules protobuf-javalite-3.11.0.jar (com.google.protobuf:protobuf-javalite:3.11.0) and protobuf-lite-3.0.1.jar (com.google.protobuf:protobuf-lite:3.0.1)

EncryptedSpirit, 2021-03-19:
Hello,

Unfortunately I didn't. Now I'm using the [flutter_blue_lib package](https://pub.dev/packages/flutter_ble_lib/install) and I don't have the issue. You should try it

DeveloperJ11, 2021-03-19:
Hi thanks

I followed your advice and tried this library and it works well, I already got the scanned devices, I have not been able to connect to the device, you will have an example of how to make this connection and be able to read a characteristic of a service.

EncryptedSpirit, 2021-03-20:
Yes of course. I'll show you an example from my code.

So after you discover a specific device you have a DEVICE object. This DEVICE object contain a future function to connect.

it should be look like this:

```
try {
 await device.peripheral.connect();
} catch (error) {
 print(error);
}
```

Once you are connected to the device you can send data.

You can discover all the services and characteristics like this.
```await _device.peripheral.discoverAllServicesAndCharacteristics();```
Store them in a variable
```List<Service> services = await _device.peripheral.services();```
And go throw each one
```
    services.forEach((service) async {
      var servChars = await service.characteristics();
      servChars.forEach((serChar) async {
         // Send data here
      });
    });
```
And send data like this
``` 
await _device.peripheral.writeCharacteristic(service.uuid, serChar.uuid, Uint8List.fromList(utf8.encode(data)), false);
```

***I don't think is the optimal way to do that. I'm beginning with BLE services and Characteristic but that work for me. Hope that was helpful.***

DeveloperJ11, 2021-03-22:
Hi! thank you very much.

You don't have the problem that the device automatically disconnects after a few seconds of connection?

D/BluetoothGatt(29955): onConnectionUpdated() - Device= interval=495 latency=0 timeout=600 status=0
D/BluetoothGatt(29955): onClientConnectionState() - status=19 clientIf=16 device=
D/BluetoothManager(29955): getConnectionState()
D/BluetoothManager(29955): getConnectedDevices
D/BluetoothGatt(29955): close()
D/BluetoothGatt(29955): unregisterApp() - mClientIf=16
I/flutter (29955): **********************************
I/flutter (29955): iTrack DESCONECTANDO.....
I/flutter (29955): **********************************

EncryptedSpirit, 2021-03-28:
No, I don't think that I have this issue

bkekelic2, 2021-04-08:
Downgrade version to:
`flutter_blue: 0.7.2`

It worked for me