pauldemarco/flutter_blue #767: Stream<BluetoothState> get state does not return a broadcast stream

anthonyorona, 2021-01-25:
Steps to Reproduce:

From any widget builder 

    import 'dart:developer' as developer;

    developer.log(
      "IS BROADCAST?: ${FlutterBlue.instance.state.isBroadcast}",
      name: 'Flutter Blue Stream',
    );

    [Flutter Blue Stream] IS BROADCAST?: false

https://github.com/pauldemarco/flutter_blue/blob/master/lib/src/flutter_blue.dart#L55

Having a broadcast stream is necessary when multiple widgets that are not under the same parent in the widget tree (e.g. body and floatingActionButton widgets) need to respond to a change in the instance state.

narritt, 2021-02-01:
How is possible to make it broadcast? I tried to add `.asBroadcastStream()` to the end of `yield*` of this getter, but it changed nothing.
Searching for a solution, it's a big problem for me.

d-wolf, 2021-04-27:
@narritt  you can work around by creating a singleton wrapper class around the bluetooth state:

```
import 'dart:async';
import 'package:flutter_blue/flutter_blue.dart';

import 'package:rxdart/rxdart.dart';

class BluetoothStateBroadcastWrapper {
  final FlutterBlue _ble = FlutterBlue.instance;

  BehaviorSubject<BluetoothState> _stateStreamController = BehaviorSubject();
  Stream<BluetoothState> get btStateStream => _stateStreamController.stream;

  StreamSubscription<BluetoothState> _sub;

  static final BluetoothStateBroadcastWrapper instance =
      BluetoothStateBroadcastWrapper._privateConstructor();

  BluetoothStateBroadcastWrapper._privateConstructor() {
    _sub = _ble.state.listen(_stateStreamController.sink.add);
  }

  void dispose() {
    _stateStreamController?.close();
    _sub?.cancel();
  }
}

```

Now you can listen to the btStateStream everywhere you like: 
```
BluetoothStateBroadcastWrapper.instance.btStateStream.listen((event) {
        
      });
```
.