pauldemarco/flutter_blue #768: Not working after building apk

praveenkprathap, 2021-01-27:
Everything works fine in debug mode. when we build the apk it will not scan for devices.

This is the permissions I used are below.
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE"/>
    <uses-feature android:name="android.hardware.bluetooth" />

akash-flutterdev, 2021-02-02:
yes i am facing same issue, build apk not working
plz address issue

akash-flutterdev, 2021-02-02:
just do this in android folder app/build.gradle

 `buildTypes {
        release {
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
        }
    }`

NaarGes, 2021-02-23:
I have the same problem
here is the log of error that happens on scan 

2021-02-24 02:49:15.840 1761-2088/? I/flutter: Error starting scan.
2021-02-24 02:49:15.840 1761-2088/? I/flutter: PlatformException(startScan, Field androidScanMode_ for b.e.a.s0 not found. Known fields are [private int b.e.a.s0.g, private b.d.d.b0$i b.e.a.s0.h, private boolean b.e.a.s0.i, private static final b.e.a.s0 b.e.a.s0.j, private static volatile b.d.d.a1 b.e.a.s0.k], java.lang.RuntimeException: Field androidScanMode_ for b.e.a.s0 not found. Known fields are [private int b.e.a.s0.g, private b.d.d.b0$i b.e.a.s0.h, private boolean b.e.a.s0.i, private static final b.e.a.s0 b.e.a.s0.j, private static volatile b.d.d.a1 b.e.a.s0.k]
        at b.d.d.v0.a(Unknown Source:72)
        at b.d.d.v0.a(Unknown Source:715)
        at b.d.d.v0.a(Unknown Source:12)
        at b.d.d.k0.a(Unknown Source:60)
        at b.d.d.k0.a(Unknown Source:49)
        at b.d.d.d1.a(Unknown Source:17)
        at b.d.d.d1.a(Unknown Source:4)
        at b.d.d.z$a.a(Unknown Source:9)
        at b.d.d.z$a.a(Unknown Source:4)
        at b.d.d.z$a.a(Unknown Source:0)
        at b.d.d.a$a.a(Unknown Source:2)
        at b.e.a.b.b(Unknown Source:10)
        at b.e.a.b.a(Unknown Source:1414)
        at f.a.c.a.j$a.a(Unknown Source:17)
    	at io.flutter.e

yang6626, 2021-02-24:
how you fix it. bro

akash-flutterdev, 2021-02-25:
ok sorry for late reply
here is solution just add in androidf/app/build.gradle

buildTypes {
    release {
        // TODO: Add your own signing config for the release build.
        // Signing with the debug keys for now, so `flutter run
--release` works.
        signingConfig signingConfigs.debug
        shrinkResources false
        minifyEnabled false
    }
}

 shrinkResources false
        minifyEnabled false

afsanchez77, 2021-03-04:
Worked for me!
Thank you for sharing.

cdExpat, 2021-03-16:
Worked for me too

Thanks alot

spekary, 2021-03-23:
See #772

biner88, 2021-03-24:
In the proguard-rules file, add a line:
 `-keep class com.pauldemarco.flutter_blue.** { *; }`

spekary, 2021-03-24:
That worked

akash-flutterdev, 2021-03-24:
thank u it's worked