pauldemarco/flutter_blue #772: not working in release

SamiraIhsan, 2021-02-04:
the prob in Stream to get the state and scan devices, it work fine on debug but in release doesn't work

bdlukaa, 2021-02-04:
Make sure you added the right permissions

SamiraIhsan, 2021-02-04:
No, I have already added the correct permissions and it was it works fine, and same case for the example existing in this plugin, and it appears in flutter 1.22.5

sornen, 2021-02-08:
Fix for this is discussed in #726

In pubspec.yaml
`
dependencies:
 flutter_blue: '0.7.2'`

or for v0.7.3 compile with
`flutter build apk --no-shrink`

SamiraIhsan, 2021-02-09:
Thanks, yes I have fixed the issue by adding
buildTypes {
        release {
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
      }
}

jimykhan, 2021-02-20:
@SamiraIhsan  same issue i am facing i try to create apk in release mode but flutter blue not working can you please guide how u fixed this issue please
where u add the following code

SamiraIhsan, 2021-02-22:
add this code in [project_name]/android/app/build.gradle
```
    buildTypes {
        release {
            signingConfig signingConfigs.release
            shrinkResources false    // this line
            minifyEnabled false       // this line
        }
    }
```

jimykhan, 2021-02-22:
thank you sir for ur response i did this but it not work for me
i am using flutter_blue 0.7.3 and flutter version 1.22.6 stable please let
me know if u have solution

meridorkobi, 2021-02-23:
I have the same issue 
Did you try with 0.7.2?

meridorkobi, 2021-02-23:
i solved the issue by adding 
--no-shrink

to my build command

jimykhan, 2021-02-23:
Yes, I did. But not working

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

jimykhan, 2021-02-23:
mDevices size: 1  brother i also face this issue when i connect device it
not show in connected device

jeroen1602, 2021-03-16:
I made another issue with some proton rules that work #784. They may be a bit too broad though.

spekary, 2021-03-23:
@pauldemarco 
Tried 0.8.0 and still seeing the problem. Building just in release mode, with none of the mods mentioned above:

```
E/flutter ( 1455): [ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: PlatformException(startScan, Field androidScanMode_ for c.c.a.y not found. Known fields are [private int c.c.a.y.f, private c.a.c.b0$i c.c.a.y.g, private boolean c.c.a.y.h, private static final c.c.a.y c.c.a.y.i, private static volatile c.a.c.a1 c.c.a.y.j], java.lang.RuntimeException: Field androidScanMode_ for c.c.a.y not found. Known fields are [private int c.c.a.y.f, private c.a.c.b0$i c.c.a.y.g, private boolean c.c.a.y.h, private static final c.c.a.y c.c.a.y.i, private static volatile c.a.c.a1 c.c.a.y.j]
E/flutter ( 1455): 	at c.a.c.v0.n0(Unknown Source:72)
E/flutter ( 1455): 	at c.a.c.v0.T(Unknown Source:697)
E/flutter ( 1455): 	at c.a.c.v0.R(Unknown Source:12)
E/flutter ( 1455): 	at c.a.c.k0.e(Unknown Source:60)
E/flutter ( 1455): 	at c.a.c.k0.a(Unknown Source:49)
E/flutter ( 1455): 	at c.a.c.d1.d(Unknown Source:17)
E/flutter ( 1455): 	at c.a.c.d1.e(Unknown Source:4)
E/flutter ( 1455): 	at c.a.c.z$a.A(Unknown Source:9)
E/flutter ( 1455): 	at c.a.c.z$a.z(Unknown Source:4)
E/flutter ( 1455): 	at c.a.c.z$a.m(Unknown Source:0)
E/flutter ( 1455): 	at c.a.c.a$a.l(Unknown Source:2)
E/flutter ( 1455): 	at c.c.a.b.q(Unknown Source:10)
E/flutter ( 1455): 	at c.c.a.b.onMethodCall(Unknown Source:1583)
E/flutter ( 1455): 	at d.a.c.a.j$a.a(Unknown Source:17)
E/flutter ( 1455): 	at io.flutter.embedding.engine.e.b.d(Unknown Source:57)
E/flutter ( 1455): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(Unknown Source:4)
E/flutter ( 1455): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/flutter ( 1455): 	at android.os.MessageQueue.next(MessageQueue.java:336)
E/flutter ( 1455): 	at android.os.Looper.loop(Looper.java:174)
E/flutter ( 1455): 	at android.app.ActivityThread.main(ActivityThread.java:7405)
E/flutter ( 1455): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 1455): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:502)
E/flutter ( 1455): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
E/flutter ( 1455): , null)
E/flutter ( 1455): #0      FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:123)
```

ayushin, 2021-04-23:
Same problem, and no-shrink doesn't solve the problem in 0.8.0

spekary, 2021-04-23:
I solved this by creating proguard rules on Android:
```
-keep class com.pauldemarco.flutter_blue.Protos* { *; }
```

Have yet to test on iOS though

ayushin, 2021-04-23:
OK so for me the combination of --no-shrink and -keep class com.pauldemarco.flutter_blue.Protos* { *; }
worked finally.

Would be great to at least document it.

d-wolf, 2021-04-27:
>             shrinkResources false    // this line
>             minifyEnabled false       // this line

> @jimykhan add this code in [project_name]/android/app/build.gradle
> 
> ```
>     buildTypes {
>         release {
>             signingConfig signingConfigs.release
>             shrinkResources false    // this line
>             minifyEnabled false       // this line
>         }
>     }
> ```

This solves the problem for me with Version 0.8.0.

alexflav23, 2021-05-19:
@MrCsabaToth How do you create proguard rules in flutter? apologies, beginner question.

jeroen1602, 2021-05-19:
@alexflav23 sure, first you will need to create the proguards rules file at  `android/app/proguard-rules.pro`. In this file add the proguard rules as described by @MrCsabaToth or in the issue #784. After this we will need to tell gradle that these rules exist.
Go into the `android/app/build.gradle` file and add the `proguardFiles` option to the correct build type. For you that would probably be:

```gradle
android {
    release {
        proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
    }
}
```

I've added them to my own project so you can copy what you need from there: https://github.com/jeroen1602/lighthouse_pm/tree/master/android/app

For more info you can check the developer documentation https://developer.android.com/studio/build/shrink-code

ertandagdelen, 2021-07-18:
> @jimykhan add this code in [project_name]/android/app/build.gradle
> 
> ```
>     buildTypes {
>         release {
>             signingConfig signingConfigs.release
>             shrinkResources false    // this line
>             minifyEnabled false       // this line
>         }
>     }
> ```

with flutter build apk --no-shrink  worked for me.

spekary, 2021-07-20:
```
minifyEnable false
shrinkResources false
```

Will turn off minifaction for your entire code base, which is probably not want you want. Yes, it will work, but it will unnecessarily increase the size of your apk file. Using the proguard rules approach will only do this for flutter-blue, and so the rest of your code will minimize.

Nurdiyor, 2021-08-21:
> the prob in Stream to get the state and scan devices, it work fine on debug but in release doesn't work

PLease add your android app proguard-rules.pro file and paste the code inside to this file.
-keep class com.pauldemarco.flutter_blue.** { *; }

DrumsnChocolate, 2021-08-24:
I just encountered the same problem, and it took me quite some time to find this thread. Any idea on when #784 will be fixed?

tijanirf, 2021-09-05:
> thank you sir for ur response i did this but it not work for me i am using flutter_blue 0.7.3 and flutter version 1.22.6 stable please let me know if u have solution
> […](#)

Strange, this solution seems to work on my application. I am also using ```flutter_blue: ^0.7.3``` and ```flutter 1.22.6```

Emadul-Hasan, 2021-10-23:
> > Issue: Not working in after build, app work properly in debug mode Hi devs, I have tried everything here, But still could not solve it. Please, someone, help me regarding this same issue. Feeling very bad after trying to solve this issue for the last 48 hours. flutter blue : ^0.8.0 flutter: ^2.2.3 Stable
> > My app: https://github.com/Emadul-Hasan/ble_splash_x
> 
> @Emadul-Hasan So many questions:
> 
> 1. I'm on stable channel, version 2.5.1 Flutter and now there's a newer version. Is there any reason (related to this issue maybe) why you are on 2.2.3?
> 2. Have you installed your production build on any of your test devices?
> 3. What error messages you see in LogCat when you install your production build on your test device(s) and reproduce your issue?
> 4. I see you have a proguard rule file in your source but it's only against the ProtoBuf classes. Does your issue reproducable if you put these two more generic rules `-keep class com.pauldemarco.flutter_blue.* { *; }` and `-keep class com.pauldemarco.flutter_blue.Protos* { *; }` instead of the ProtoBuf one? You don't need `minifyEnable false` or `shrinkResources false` when you try these.
> 
> Please provide answers to these questions.

1. There is no specific reason to not updating the flutter version
2. I have installed the realsed apk to my device. till scanning is working fine after minifying but data is not being received and application freezes. But in debug mode (In the same device) it working fine without any issue, there is not error shown.
3. I did not used emulator for testing, I just used my real device for testing the apks. So i did not observed in longcat.
4. I have tried both the way, using both of the proguard rules, but unfortunately, it did not work for me.

jenow, 2022-10-25:
> @alexflav23 sure, first you will need to create the proguards rules file at `android/app/proguard-rules.pro`. In this file add the proguard rules as described by ~@MrCsabaToth~ @d-wolf or in the issue #784. After this we will need to tell gradle that these rules exist. Go into the `android/app/build.gradle` file and add the `proguardFiles` option to the correct build type. For you that would probably be:
> 
> ```groovy-gradle
> android {
>     release {
>         proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
>     }
> }
> ```
> 
> I've added them to my own project so you can copy what you need from there: https://github.com/jeroen1602/lighthouse_pm/tree/master/android/app
> 
> Edit: here is the same link at a specific commit in case it changes in the future and someone still needs help https://github.com/jeroen1602/lighthouse_pm/tree/10eeccbd2200f82dbbff5dee27c18f76ea9151ea/android/app
> 
> For more info you can check the developer documentation https://developer.android.com/studio/build/shrink-code

This should definitely be part of the documentation..