pauldemarco/flutter_blue #777: Unable to read data from characteristic on Android but able to read on iOS

vaibhiarora03, 2021-02-18:
I am unable to read data from characteristic on Android but have been able to read the data on iOS device. Can anyone tell me what I'm missing so that data can be read on Android too.

**Connecting the device:-**
`
try {
      await device.connect();
    } catch (error) {
      if (error.code != 'already_connected') {
        throw error;
      }
    } finally {
      List<BluetoothService> deviceServices = await device.discoverServices();
       _examineDeviceServicesForTestConnection(deviceServices, device);
    }
`

**Examining the service with characteristic read and notify:-**

`
_examineDeviceServicesForTestConnection(
      List<BluetoothService> services, BluetoothDevice device) async {
    BluetoothCharacteristic testReadCharacteristic;
    BluetoothCharacteristic testWriteCharacteristic;
    serviceLoop:
    for (BluetoothService service in services) {
      if (service.characteristics.isNotEmpty) {
        characteristicLoop:
        for (BluetoothCharacteristic characteristic
            in service.characteristics) {
          if (characteristic.properties.notify) {
            testReadCharacteristic = characteristic;
          }
          if (characteristic.properties.write) {
            testWriteCharacteristic = characteristic;
          }
        }
        if (testReadCharacteristic != null && testWriteCharacteristic != null) {
          final testCommand = "TEST_COMMAND";
          final convertedTestCommand = utf8.encode(testCommand);

          //reading data from characteristics
          _readDataFromCharacteristics(
              testReadCharacteristic, testWriteCharacteristic, device);

          //write command to device
          await testWriteCharacteristic.write(convertedTestCommand);
        }
      }
    }
  }
`

**Reading the data from characteristic**
`
_readDataFromCharacteristics(BluetoothCharacteristic readerCharx,
      BluetoothCharacteristic writerCharx, BluetoothDevice device) async {
    await readerCharx.setNotifyValue(true);
    readerCharx.value.listen((response) {
      var decodedResponse = utf8.decode(response);
      print(decodedResponse);
    });
  }
`

The data is successfully received when I use this on the iOS device. The device with which I'm connecting to is an BLE device. Can you please help me pointing where I went wrong?

Thanks

MuFengYi, 2021-11-05:
Did you solve it, I also encountered this problem.

vaibhiarora03, 2021-11-05:
@MuFengYi No. For android we had classic devices. So used Flutter_blue_serial instead.