pauldemarco/flutter_blue #778: Convert to a federated plugin

robert-ancell, 2021-02-21:
To add [Linux support](https://github.com/pauldemarco/flutter_blue/issues/330) this will be easier if *flutter_blue* is a [federated plugin](https://flutter.dev/docs/development/packages-and-plugins/developing-packages#federated-plugins). The Linux support can all be done in Dart code using [bluez](https://pub.dev/packages/bluez), which is not possible with the current *flutter_blue* plugin structure.

robert-ancell, 2021-02-21:
@pauldemarco I'm happy to do this work - is this something that you want in this project?

spekary, 2021-02-26:
I think this would be a great. In fact, anything that would lighten the load. @pauldemarco perhaps its time to move this to a group managed project. Bluetooth is critical infrastructure to many apps, and with the advent of desktop platforms, constant changes in OS's and the bluetooth spec, null safety, etc. I can't imagine how one person could keep up.

robert-ancell, 2021-03-09:
I've started working on Linux support in [flutter_reactive_ble](https://pub.dev/packages/flutter_reactive_ble) and am no longer looking at flutter_blue at this time.

gocal, 2021-03-25:
Hi @robert-ancell we're actively working on it - the project has been successfully converted to support federated plugins, and we added linux model as well - we can check current state here https://github.com/allbrightio/flutter_blue/tree/federated_plugin