pauldemarco/flutter_blue #779: Android 10+: after application minimizing only first scan found devices

narritt, 2021-02-23:
Tested on Samsung Galaxy S20 (Android 11); S9+ (Android 10).

Make two delayed scans like this:
- delay 2 seconds (to minimize the application)
- scan 4 seconds
- delay 1 second
- scan 4 seconds

Trigger this scenario and immediately minimize the application.

In the project example I wrote this code on the floating button tapping:
```
          print("Minimize application now!");
          await Future.delayed(Duration(seconds: 2));
          print("Starting first scan...");
          int counter = 0;
          FlutterBlue.instance.scan(
            timeout: Duration(seconds: 4),
            allowDuplicates: false
          ).listen((event) => counter++);
          await Future.delayed(Duration(seconds: 5));
          print("Found $counter devices");
          counter = 0;
          print("Starting another scan...");
          FlutterBlue.instance.scan(
              timeout: Duration(seconds: 4),
              allowDuplicates: false
          ).listen((event) => counter++);
          await Future.delayed(Duration(seconds: 5));
          print("Found $counter devices");
          return;
```
On the first scan, it found 57 devices, but all next scans - zero:
```
I/flutter ( 1740): Minimize application now!
I/ViewRootImpl@caf443a[FlutterActivity]( 1740): ViewPostIme pointer 0
I/ViewRootImpl@caf443a[FlutterActivity]( 1740): MSG_WINDOW_FOCUS_CHANGED 0 1
D/FlutterBluePlugin( 1740): mDevices size: 0
I/flutter ( 1740): AppLifecycleState.inactive
I/flutter ( 1740): AppLifecycleState.paused
I/flutter ( 1740): Starting first scan...
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothLeScanner( 1740): Start Scan with callback
D/BluetoothLeScanner( 1740): onScannerRegistered() - status=0 scannerId=7 mScannerId=0
D/FlutterBluePlugin( 1740): mDevices size: 0
D/FlutterBluePlugin( 1740): mDevices size: 0
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothLeScanner( 1740): Stop Scan with callback
D/FlutterBluePlugin( 1740): mDevices size: 0
I/flutter ( 1740): Found 57 devices
I/flutter ( 1740): Starting another scan...
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothLeScanner( 1740): Start Scan with callback
D/BluetoothLeScanner( 1740): onScannerRegistered() - status=0 scannerId=7 mScannerId=0
D/FlutterBluePlugin( 1740): mDevices size: 0
D/FlutterBluePlugin( 1740): mDevices size: 0
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothAdapter( 1740): STATE_ON
D/BluetoothLeScanner( 1740): Stop Scan with callback
I/flutter ( 1740): Found 0 devices
```

vaimikpatel2908, 2021-02-26:
I am facing the same problem while scanning repeatedly using Timer.Periodic(). My timer Duration is 7 seconds and a scan duration is 5 seconds still facing the same issue.

jeroen1602, 2021-03-16:
This one is probably related #772

vaimikpatel2908, 2021-03-16:
It might be possible. According to my view, it was not working in the debug mode. Therefore it might not be as well. The strange thing is that it started working after restarting my testing device but didn't help by restarting editors or executing command flutter clean. Therefore, there might be something wrong with the plugin(a small thing which might went wrong) or it just closes the stream or not attaching the proper callback. But, It looks like the rare thing, as I could not replicate that issue again exactly.