pauldemarco/flutter_blue #783: Connection error: [onConnectionStateChange] status: 133 newState: 0

sami-samcla, 2021-02-26:
- **flutter_blu** version: **0.7.3** (also test it with git repo defined into dependecies in pubspec.yaml with same result)
- **Problem detected on**: Samsung Galaxy J6 (Android 10 - API 29) and Huawei P8 Lite (Android 7.0 API 24). Error are more frequently in Samsung than Huawei.

Flutter Doctor info:
[✓] Flutter (Channel dev, 1.27.0-4.0.pre, on macOS 11.1 20C69 darwin-x64, locale en-GB)
    • Flutter version 1.27.0-4.0.pre at /Users/sissa/flutter
    • Framework revision f8cd24de95 (10 days ago), 2021-02-16 11:24:17 -0800
    • Engine revision 1d537824d6
    • Dart version 2.13.0 (build 2.13.0-30.0.dev)

[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at /Users/sissa/Library/Android/sdk
    • Platform android-30, build-tools 29.0.2
    • ANDROID_HOME = /Users/sissa/Library/Android/sdk
    • Java binary at: /Applications/Android Studio 2.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6915495)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.4, Build version 12D4e
    • CocoaPods version 1.10.0

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 3.5)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)

[✓] Android Studio (version 4.1)
    • Android Studio at /Applications/Android Studio 2.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6915495)

[✓] VS Code (version 1.40.0)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension can be installed from:
      🔨 https://marketplace.visualstudio.com/items?itemName=Dart-Code.flutter

[✓] Connected device (3 available)
    • SM J610FN (mobile)       • bbc7a841                                 • android-arm    • Android 10 (API 29)
    • SAMCLA’s iPhone (mobile) • f063d3c4cfd212cac4386163446403a522fa40cb • ios            • iOS 14.4
    • Chrome (web)             • chrome                                   • web-javascript • Google Chrome 88.0.4324.192

This error occurs on connection action. Scan works correctly. iOS not tested.
Note that: location permissions has granted. 

I have tried solve the problem with this [solution](https://github.com/pauldemarco/flutter_blue/issues/231), but not luck.

This is my connection code:

`@override
  Future connect({int timeoutDurationInSeconds, Function onTimeOut, Function onConnected, bool stopScanProcess = true}) async {
    try {
      if (stopScanProcess) {
        scanTimeOut?.cancel();
        stopScan();
      }

      resetCharacteristics();

      isConnecting = true;
      //checkState();
      await checkCurrentState();

      await deviceToConnect
          .connect(autoConnect: false)
          .timeout(
            Duration(seconds: kBleTimeoutConn),
            onTimeout: () => onTimeOut(),
          )
          .then(
        (value) async {
          await checkCurrentState();

          if (deviceState == BluetoothDeviceState.connected) {
            connectionSetup(onConnected: onConnected);
          }
        },
      );
    } on Exception catch (e) {
        isConnecting = false;

        print(e);
        throw BluetoothConnectionException();
    }
  }

  @override
  Future<void> checkCurrentState() async {
    if (null != deviceToConnect) {
      await for (BluetoothDeviceState value in deviceToConnect.state) {
        deviceState = value;
        
        return deviceState;
      }
      return null;
    } else
      return null;
  }

@override
  Future disconnect({Function onDisconnected}) async {
    if (null != deviceToConnect) {
      stopScan();

      resetCharacteristics();

      print('disconnecting ${deviceToConnect?.name}');

      await deviceToConnect.disconnect();
      print('************ disconnected');

      isConnecting = false;

      if (null != deviceBluetoothResponseSubscription) deviceBluetoothResponseSubscription.cancel();

      await Future.delayed(Duration(milliseconds: 500));
      //checkState();
      //await checkCurrentState();
      deviceState = BluetoothDeviceState.disconnected;
    }
    if (null != onDisconnected) onDisconnected();
  }
`

Dhi13man, 2021-02-28:
133 is Device Not Found GATT status. Ensure that the other device is advertising BLE services.

sami-samcla, 2021-03-01:
@Dhi13man I'm sure that the device is advertising BLE services correctly because iOS devices works fine.

sami-samcla, 2021-03-03:
@pauldemarco do you have any suggestion for this problem? I'm stucked from days.

sami-samcla, 2021-06-22:
Upgraded version to 0.8.0 and problem is not solved. I can't believe that I the only one with this problem. The connection behavior in Android devices it's very disappointing. It's possible this problem comes from a bug in my implementation, but I need more ideas to solve it.

RafaRuiz, 2021-06-27:
I'm having this problem too

wangkaibule, 2021-08-30:
Here's a discussion ([link](https://devzone.nordicsemi.com/f/nordic-q-a/33313/android-gatt-133-error)) suggests a known BUG in Android implementation: In short, Android sometimes incorrectly terminate the previous connection session, leaving the remote device in wrong state.

Maybe what we need is a way to ensure the disconnection procedure has correctly committed. However, we currently cannot achieve this purpose in flutter_blue, because a `disconnect` command has no effect on the device which has already been marked as "disconnected".