pauldemarco/flutter_blue #787: NNBD Support

navaronbracke, 2021-03-10:
As of March 3rd 2021 NNBD is stable along with the new major version 2.0.0 of the Flutter framework.

This plugin should migrate to NNBD.

DokySp, 2021-03-11:
got a same issue. I already did migrate my project and always use `$ flutter run --no-sound-null-safety` and it works, but I think this is not a proper solution of this issue.

navaronbracke, 2021-03-11:
@pauldemarco I'd like to work on the migration myself but I don't have a Linux machine, so I don't know if I could update the proto files.

qurm, 2021-03-12:
I think Paul, understandably, has not been able to find enough time to maintain this.  If @navaronbracke or anyone else want to make a fork and continue to develop towards NNBD, I'd be a willing contributor.  It looks like there are a number of good pull requests already for other issues, so there are other potential contributors.

Or @pauldemarco - would you consider involving others to maintain this library, or help with issues etc?  Let us know your plans.

qurm, 2021-03-16:
Thanks @pauldemarco  - I see you are doing some work on Null safety - much appreciated.  Let us know if you need some testing or other assistance.

navaronbracke, 2021-03-16:
Version 0.8.0 now supports Null Safety. Closing.

closed 2021-03-16 by navaronbracke

MrCsabaToth, 2021-03-25:
I'm trying to Pug Get and I'm running into "Because every version of flutter_driver from sdk depends on convert 2.1.1 and every version of flutter_blue depends on convert ^3.0.0, flutter_driver from sdk is incompatible with flutter_blue.". I'm on Flutter beta channel, wouldn't want to go dev. How can I overcome this?

```
flutter doctor -v
[✓] Flutter (Channel beta, 2.0.2, on Linux, locale en_US.UTF-8)
    • Flutter version 2.0.2 at /home/csaba/flutter
    • Framework revision 8962f6dc68 (2 weeks ago), 2021-03-11 13:22:20 -0800
    • Engine revision 5d8bf811b3
    • Dart version 2.12.1

[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Android SDK at /home/csaba/Android/Sdk
    • Platform android-30, build-tools 30.0.3
    • Java binary at: /home/csaba/android-studio/jre/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)
    • All Android licenses accepted.

[✓] Chrome - develop for the web
    • Chrome at google-chrome

[✓] Android Studio
    • Android Studio at /home/csaba/android-studio
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)

[✓] VS Code (version 1.54.3)
    • VS Code at /usr/share/code
    • Flutter extension version 3.20.0

[✓] Connected device (2 available)
    • ONEPLUS A6003 (mobile) • c98b8376 • android-arm64  • Android 10 (API 29)
    • Chrome (web)           • chrome   • web-javascript • Google Chrome 89.0.4389.90

• No issues found!
```

navaronbracke, 2021-03-26:
I can run `flutter pub get` with both the Flutter Driver and `flutter_blue` (version 0.8.0) successfully. I am also running on the beta channel (on stable the flutter driver indeed has convert 2.1.1 as dependency). Did you run `flutter channel beta` and `flutter upgrade` afterwards? 

```
[√] Flutter (Channel beta, 2.1.0-12.2.pre, on Microsoft Windows [Version 10.0.19041.867], locale nl-BE)
    • Flutter version 2.1.0-12.2.pre at E:\Flutter\flutter
    • Framework revision 5bedb7b1d5 (8 days ago), 2021-03-17 17:06:30 -0700
    • Engine revision 711ab3fda0
    • Dart version 2.13.0 (build 2.13.0-116.0.dev)

```

MrCsabaToth, 2021-03-26:
> I can run `flutter pub get` with both the Flutter Driver and `flutter_blue` (version 0.8.0) successfully. I am also running on the beta channel (on stable the flutter driver indeed has convert 2.1.1 as dependency). Did you run `flutter channel beta` and `flutter upgrade` afterwards? `flutter channel beta` on its own does not update your local Flutter sources. Thats what `flutter upgrade` is 

Well, flutter usually is very keen on telling me when an update is available. This time it kept silent, I was able to upgrade one more, I guess things are changing quickly partially to post 2.0 stable release reasons and partially to continued null safety. So this particular issue got solved, but overall I had to revert back my changes and now even the code which was running last night fine is failing with 3 deep transient dependency problems. It's not flutter_blue any more though.