pauldemarco/flutter_blue #788: Bluetooth adapter not available on MacOS

divan, 2021-03-11:
Hi,

trying to run an example app on Macbook Pro M1 and getting Bluetooth state = null and "Bluetooth adapter not available" screen/message.

Any idea what can be wrong? I've tried to put iOS bluetooth related strings into macos/Runner/Info.plist, but no luck. Also added manuall debug build to the list of bluetooth-permitted apps in Settings->Security/Privacy MacOS screen.

I don't see any errors or messages in the logs, so not sure what can be wrong here. Any suggestions?

<details>
<summary>flutter doctor -v output</summary>
<pre>✗ flutter doctor -v
[✓] Flutter (Channel stable, 2.0.1, on macOS 11.2.2 20D80 darwin-arm, locale en-GB)
    • Flutter version 2.0.1 at /Users/divan/flutter
    • Framework revision c5a4b4029c (7 days ago), 2021-03-04 09:47:48 -0800
    • Engine revision 40441def69
    • Dart version 2.12.0

[!] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Android SDK at /Users/divan/Library/Android/sdk
    • Platform android-30, build-tools 30.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6915495)
    ! Some Android licenses not accepted.  To resolve this, run: flutter doctor --android-licenses

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.4, Build version 12D4e
    • CocoaPods version 1.10.0

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 4.1)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6915495)

[✓] VS Code (version 1.54.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.20.0

[✓] Connected device (2 available)
    • macOS (desktop) • macos  • darwin-arm64   • macOS 11.2.2 20D80 darwin-arm
    • Chrome (web)    • chrome • web-javascript • Google Chrome 88.0.4324.192

! Doctor found issues in 1 category.</pre>
</details>

divan, 2021-03-11:
Found it:
 - open macos/Runner.xcworkspace (it'll open in XCode)
 - find target "Runner" and go to the Signing & Capabilities tab
 - in 'App Sandbox' find and enable "Hardware" / "Bluetooth" checkbox
 - rebuild app

PS. It still shows "bluetooth adapter is unavailable" after granting permissions. Restart helps (I think there's similar issue for iOS)

closed 2021-03-11 by divan

areille, 2021-03-11:
Maybe we could add a macos-specific setup section in README, as there is only android and iOS for now.

Something like :

#### MacOS

In the __macos/Runner/DebugProfile.entitlements__ and __macos/Runner/Release.entitlements__ let's add:
```
<key>com.apple.security.device.bluetooth</key>
<true/>
```