pauldemarco/flutter_blue #789: Null safe support

tanga13970, 2021-03-12:
Flutter has published flutter 2 which requires all dependencies update to Null safe.

Can flutter blue update to Null safe?

dspells, 2021-03-12:
I've been using flutter_blue for months with the beta version of flutter to get the null safety feature
I just updated my project to use flutter 2 and all but 2 of my packages dependencies were updated to use 'sound null safety'.

I would like to know if 'null safety' is in the plans and a timeline if you have one.

Here is one of the warnings that flutter puts out

info: The library 'package:flutter_blue/flutter_blue.dart' is legacy, and should not be imported into a null safe library. (import_of_legacy_library_into_null_safe

Thanks in advance for any information you have on your plans for null safety

navaronbracke, 2021-03-16:
There is a new version, 0.8.0, that supports Null Safety.

dspells, 2021-03-18:
I am using 0.8.0 in my app and it is no longer giving warnings and I have had no problems so far. Thanks.

tanga13970, 2021-03-23:
Thanks for the quick response to Null Safty support