pauldemarco/flutter_blue #790: Bluetooth adapter is unavailable IOS

h582618, 2021-03-13:
Having an issue regarding bluetooth. When i launch the app i get no errors but just a message that states "Bluetooth adapter is unavailable".  
The bluetooth state seems to be null.

What i have tried so far:

Added the NSBluetoothAlwaysUsageDescription etc in ios/Runner/Info.plist.
Tried version 0.7.2 and 0.5.3.
Tried an earlier version of flutter.
Also started a new project with only the example code. Still same issue.

Any idea what can be wrong?

navaronbracke, 2021-03-13:
Wat is the iOS version you are running on? I'd suggest iOS 8 or higher, as that is the requirement for the API's. Does the device actually have bluetooth? (Simulators do not have access to Bluetooth I think) Lastly, there are other keys in Info.plist that might be needed for Bluetooth, maybe you forgot one? (E.g. `NSBluetoothPeripheralUsageDescription`)

h582618, 2021-03-13:
Ios version 14, i also added all the other keys for info.plist. 

But the issue seems to be that i was using the simulator. Thank you for good help.

closed 2021-03-13 by h582618